Julius2342/pyvlx #2: Roller shutter warning

tikismoke, 2017-06-26:
`
WARNING: Could not parse product: Roller shutter
`

How can i help you with this to test?
I can only grab window 0 and 1 but i have 4 device 2 shuutter and 2 window.

tikismoke, 2017-06-26:
I print the json anser of the klf200 i got:
`[{'name': 'Volet roulant cour', 'id': 0, 'scenes': ['Fermer volet cour', 'Ouvrir volet cour'], 'category': 'Roller shutter', 'typeId': 2, 'subtype': 0}, {'name': 'Fenêtre cour', 'id': 1, 'scenes': ['Fermer fenetre cour', 'Ouvrir fenetre cour'], 'category': 'Window opener', 'typeId': 4, 'subtype': 1}, {'name': 'Fenêtre jardin', 'id': 2, 'scenes': ['Fermer fenetre jardin', 'Ouvrir fenetre jardin'], 'category': 'Window opener', 'typeId': 4, 'subtype': 1}, {'name': 'Volet roulant jardin', 'id': 3, 'scenes': ['Fermer volet jardin', 'Ouvrir Volet jardin'], 'category': 'Roller shutter', 'typeId': 2, 'subtype': 0}]
`

Julius2342, 2017-06-26:
Hello! I don't have a roller shutter :-)

Probably we have to add a class named "RollerShutter" derived from Device (Just copy Window) ...

And, what reminds me: we have to add unit tests for parsing results...

Julius2342, 2017-06-26:
May you test this?

tikismoke, 2017-06-26:
Not this evening, I'll try in the week.

Did you find any way to get position? Or other API functionality?

Julius2342, 2017-06-26:
No, did not really try to find other api calls. But i could a security problem: https://gist.github.com/Julius2342/6282ded9f527e762ea50f42c2c439a1a

tikismoke, 2017-06-27:
Ok the roller-shutter class is working for me.

Julius2342, 2017-06-27:
merged to master and deleted remote branch.

Question for me is: what are the other typeids? But we probably have to ask Velux. :)

Do you also experience these stange timeout problems?

closed 2017-06-27 by Julius2342

tikismoke, 2017-06-28:
Not sure Velux will be very helpfull but hy not have a try.