Julius2342/pyvlx #7: got timeout when addressing klf200

dirk777, 2018-05-16:
**Issue:**
klf200 is responding after scene has completed, which means that the http request configured with 10s timeout will stop and raise an error, if scene takes more than 10s

**Solution:**
increase timeout or make it configurable
In my fork I already did this: https://github.com/dirk777/pyvlx/commit/6cdede0d913a81bd7a56c15b03de3ca17c6d7158
If you want, I can make another pull request out of it?

Julius2342, 2018-05-16:
PR would be highly appreciated!

Julius2342, 2018-11-29:
Hi @dirk777 : I will reject this issue. The new API/implementation does not have this problem, bc the communication works asynchronously.

Julius2342, 2018-11-29:
btw, it would be great, if you could test the new API :-)

closed 2018-11-29 by Julius2342