Julius2342/pyvlx #10: new api

tikismoke, 2018-10-12:
Hi some of my friend told me about a new documented API.

Did you have a look at: 
https://www.velux.com/api/klf200

Julius2342, 2018-10-13:
Yes i have seen it. Will consider implementing it :-)

tikismoke, 2018-10-13:
Nice to hear :)

Really interested by some value I saw in doc like current position, but not sure it's really available.

Julius2342, 2018-11-01:
I'm installing the new firmware right now. *fingers-crossed*

Julius2342, 2018-11-01:
Problem1: the webinterface is only reachable via wlan, not via LAN. (I can ping the host and i can connect to their management port 51200, but not to the webinterface ...)

Problem 2: their examples dont work :-(

Julius2342, 2018-11-03:
@tikismoke : i played around with the new api: https://github.com/Julius2342/pyvlx/tree/master/new_api

The bad news is, that my windows don't propagate the current position :-/

tikismoke, 2018-11-03:
Ok and the good news?

Does it works as except opening/closing?

I have to update the firmware first.
Then I can start some test.

Julius2342, 2018-11-03:
At least the KLF 200 is talking to me :) 

I can read scenes and read devices. At least in an example i was able to trigger a scene, but i did not implement it yet.

Julius2342, 2018-11-03:
Still a lot work to do :)

tikismoke, 2018-11-03:
Ok I'll wait a bit before start asking lot of things :)

Nice work by the way

tikismoke, 2018-11-24:
Cool see the new release. I'll try to test it this weekend.

Still no way to get current position before moving it?

Julius2342, 2018-11-25:
@dirk777 @tikismoke @cellerich @Arduous @esevend : I did a complete rewrite of PyVLX for support of the new Firmware of velux. It would be great if you could try it out. (if you upgrade the firmware, you can no longer use the old lib, it is not backwards compatible.)

@mischroe : saw that you wrote an api library for javascript. Were you able to read the current position of windows? My KLF 200 is always answering with "unknown position".

MiSchroe, 2018-11-26:
@Julius2342 : Have you tried to enable the House Monitor? After that you'll get notifications from time to time and the interface seems to know about the current status, too.

See below a dump of the notification after sending a GW_GET_NODE_INFORMATION_REQ for one of my windows:
````
2018-11-26 11:34:37: Frame GW_GET_NODE_INFORMATION_NTF
2018-11-26 11:34:37:   ActuatorAliases: (2) [[{"AliasType":55298,"AliasValue":25600},{"AliasType":55299,"AliasValue":47616}]]
2018-11-26 11:34:37:   NodeID: 4
2018-11-26 11:34:37:   Order: 4
2018-11-26 11:34:37:   Placement: 4
2018-11-26 11:34:37:   Name: Fenster Büro
2018-11-26 11:34:37:   Velocity: 0
2018-11-26 11:34:37:   ActuatorType: 4
2018-11-26 11:34:37:   ActuatorSubType: 1
2018-11-26 11:34:37:   ProductType: 54535
2018-11-26 11:34:37:   NodeVariation: 0
2018-11-26 11:34:37:   PowerSaveMode: 1
2018-11-26 11:34:37:   SerialNumber: S6'&/
2018-11-26 11:34:37:   OperatingState: 5
2018-11-26 11:34:37:   CurrentPosition: 51200
2018-11-26 11:34:37:   TargetPosition: 51200
2018-11-26 11:34:37:   FunctionalPosition1CurrentPosition: 0
2018-11-26 11:34:37:   FunctionalPosition2CurrentPosition: 63487
2018-11-26 11:34:37:   FunctionalPosition3CurrentPosition: 63487
2018-11-26 11:34:37:   FunctionalPosition4CurrentPosition: 63487
2018-11-26 11:34:37:   RemainingTime: 0
2018-11-26 11:34:37:   TimeStamp: Wed Jan 11 2012 22:05:44 GMT+0100 (Mitteleuropäische Zeit)
````

BTW: Do you have any idea how to interpret the serial number?

Some other findings I had made:

- GW_GET_LIMITATION_STATUS_REQ doesn't work. I will alway get an invalid frame error. I have tried with several different frame lengths.
- GW_MODE_SEND_NTF and GW_ACTIVATE_PRODUCTGROUP_NTF are in the list of gateway commands but not documented and not received after sending their corresponding request command.

Julius2342, 2018-11-26:
hi @MiSchroe !

thank you! I will try out house monitor!

With regards to serial number, i print it out as hex e.g. `A1:B2:C3:D4:E5:F6:A6:B8` via `":".join("{:02x}".format(c) for c in self._serial_number)`but i have no idea if this is correct.

Talking about findings, did you notice that other the documentation suggests, the build number is included within [GW_GET_NODE_INFORMATION_NTF](https://github.com/Julius2342/pyvlx/blob/master/pyvlx/frames/frame_get_node_information.py#L115).

So far, i only implemented the commands i needed to control my windows :-). Will proceed with the others asap.

MiSchroe, 2018-11-26:
Hi @Julius2342 ,

I think, that the documentation for GW_GET_ALL_NODES_INFORMATION is wrong and that there is no BuildNumber at all. Maybe, this could be your issue.

Here is what I get for both message type for the same window (raw):
````
2018-11-26 11:34:36: Data received: c0:00:7f:02:04:04:00:04:04:46:65:6e:73:74:65:72:20:42:c3:bc:72:6f:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:01:01:d5:07:00:01:1e:53:36:27:26:10:2f:00:81:05:c8:00:c8:00:00:00:f7:ff:f7:ff:f7:ff:00:00:4f:0d:f9:a7:02:d8:02:64:00:d8:03:ba:00:00:00:00:00:00:00:00:00:00:00:00:00:fb:c0
2018-11-26 11:34:36: Frame GW_GET_ALL_NODES_INFORMATION_NTF

2018-11-26 11:34:37: Data received: c0:00:7f:02:10:04:00:04:04:46:65:6e:73:74:65:72:20:42:c3:bc:72:6f:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:01:01:d5:07:00:01:1e:53:36:27:26:10:2f:00:81:05:c8:00:c8:00:00:00:f7:ff:f7:ff:f7:ff:00:00:4f:0d:f9:a8:02:d8:02:64:00:d8:03:ba:00:00:00:00:00:00:00:00:00:00:00:00:00:e0:c0
2018-11-26 11:34:37: Frame GW_GET_NODE_INFORMATION_NTF
````

The only difference is the message type, the time stamp and the CRC byte. Both message come with a length of 0x7F = 127 bytes.

Julius2342, 2018-11-27:
Hi @MiSchroe 

I added a unit test with your packets and it at least [looks sane](https://github.com/Julius2342/pyvlx/blob/master/test/frame_node_information_mischroe_test.py). While doing so i found several bugs, so thank you very much! :-)

So, what else then the build number should be this byte at [position 75](https://github.com/Julius2342/pyvlx/blob/master/pyvlx/frames/frame_get_node_information.py#L146)? (I start counting within the payload).

Thank you very much for your help!

Julius

Arduous, 2018-11-28:
Hello @Julius2342 et all. Many thanks for your work and heads up! Not sure when I'll be able to upgrade the FW of the KLF, but I will definitely feedback.

bramstroker, 2018-11-29:
@Julius2342 Thanks for updating the library to support the latest firmware. I have installed the firmware and using your updated module now. everything is working as expected. Will do some more testing..
One issue however totally unrelated to your python library, I cannot reach the web interface through LAN after the firmware update, any ideas?

pilehave, 2018-11-29:
> @Julius2342 Thanks for updating the library to support the latest firmware. I have installed the firmware and using your updated module now. everything is working as expected. Will do some more [...]

The interface can't be opened via LAN with the new firmware, only WLAN.

Julius2342, 2018-11-29:
> The interface can't be opened via LAN with the new firmware, only WLAN.

i think Velux wanted to force me to implement the new API :-)

Julius2342, 2018-11-29:
short update on the progress:

the PyVLX is now switching on house-monitor-mode (which is needed to monitor positions of windows). It is also transmitting the current (unix) time (the KLF 200 seems to forget the time after reboot). The missing piece is the updating the internal position within windows, to allow asycronous access e.g. from within HASS.

apeeters, 2018-12-01:
Works like a charm, thanks for the excellent work!

WoodsterDK, 2018-12-03:
Hi,

Regarding maintaining the actual position within the class:

After an operation has been performed a GW_COMMAND_RUN_STATUS_NTF should be sent with the byte "StatusReply = COMMAND_COMPLETED_OK".
When this is OK - you can then request the GW_GET_NODE_INFORMATION_REQ which will return with a GW_GET_NODE_INFORMATION_CFM containing the actual position.

As a sidenote you might want to add the following node type, which is a wall power socket thing:

class OnOffSwitch(OpeningDevice):

if frame.node_type == NodeTypeWithSubtype.ON_OFF_SWITCH:
   return OnOffSwitch(pyvlx=pyvlx,node_id=frame.node_id, name=frame.name)

tikismoke, 2018-12-03:
First asking and still not find a way to implement and test it :( shame on me @Julius.

Pretty sure it will be perfect when I'll find time to test it.

Julius2342, 2018-12-15:
@WoodsterDK : my system sends automatically a FrameNodeStatePositionChangedNotification after the nodes have reached a new position. I dont have to request an additional GW_GET_NODE_INFORMATION_REQ.  May you verify if this is the case on your system? (I also updated the HASS module, may you also test?).

An On-Off switch would be a different device, not derived from OpeningDevice but from Node. May I ask which product you have in use?

Julius2342, 2018-12-17:
Here is the [Pull Request](https://github.com/home-assistant/home-assistant/pull/18738) within HASS

WoodsterDK, 2018-12-17:
> @WoodsterDK : my system sends automatically a FrameNodeStatePositionChangedNotification after the nodes have reached a new position. I dont have to request an additional GW_GET_NODE_INFORMATION_REQ. [...]

@Julius2342 Super - I will give it a go - I didn't see you used the notification from the House Status Monitor.

The ON/OFF switch is a VELUX ADAPTER (KRD 100) - 0% equals off -> 100% equals on.

apeeters, 2019-01-13:
I tested the PR branch in HASS, works perfectly. Thanks for the great work! Can't wait to have it merged :)

Julius2342, 2019-01-18:
@apeeters : May I ask you to comment this within the PR? :)

pilehave, 2019-01-24:
Does someone have screenshots/examples on operation in Home Assistant? If I want to test this in a non-HASS install (manual install on Haspbian), what should I do?

Julius2342, 2019-01-29:
@pilehave :  Follow the steps to set up a dev environment: https://developers.home-assistant.io/docs/en/development_environment.html with this small modification:

```
git clone https://github.com/Julius2342/home-assistant.git
cd home-assistant
git checkout new-velux-api
python3 -m venv venv
source venv/bin/activate
./script/setup
hass
```

But i hope my branch gets merged soon *fingerscrossed*

Julius2342, 2019-02-12:
Let's close the issue. If any problems arise with the new API, please open a new issue.

closed 2019-02-12 by Julius2342