Julius2342/pyvlx #15: Unable to establish a successful connection

l-mb, 2019-02-22:
Hi, thanks for pyvlx. I'm trying to use it with Home Assistant and my KLF 200.

This is using pyvlx 0.2.8 and Python 3.7.

Unfortunately, I'm not having much luck. I double and triple checked the hostname and password, but all I'm getting is
pyvlx.exception.PyVLXException: <PyVLXException description="Unable to login" />

(I tried resetting the password to something impossible to get wrong, but that didn't help either.)

I see a connection to the 192.168.2.180:51200 being established, and data flowing back and forth. So it seems the LAN port is active correctly. Unfortunately, no details as to why the login fails.

(I've even double-checked the MAC address, that's indeed the Velux device.)

The device is running the newest 0.2.0.0.71.0 firmware (that's what it shipped with, and there's no update).

I'd really appreciate any hint on how to debug this further. Am I missing something obvious?

l-mb, 2019-02-23:
Consider my forehead smacked and sorry for being dumb.

Since all examples (even the API demo code from Velux) used the "velux123" "velux1234" password, I thought this was the web login password. What it actually wants is the same password as for connecting to the wifi.

I'll submit a PR to the Home Assistant docs to clarify this.

closed 2019-02-23 by l-mb

Julius2342, 2019-02-23:
Oh great :) i was not aware of this. The bad news is that there is no good log output to tell the password is wrong.