Julius2342/pyvlx #17: nodes.py - TypeError

gibman, 2019-03-11:
Hi running hass.io 0.89.1 with a klf 200 unit with the latest firmware.
(I just bought the unit :=)

I have various products setup in the velux webinterface.
* windows
* shutters
* blinds
* awning blinds

It seems it wont load up properly in hass.io.

hass.io configuration.yaml

velux:
  host: "192.168.1.123"
  password: "wifipass"

hass.io logs:

[homeassistant.setup] Error during setup of component velux
Traceback (most recent call last):
File “/usr/local/lib/python3.7/site-packages/homeassistant/setup.py”, line 151, in _async_setup_component
hass, processed_config)
File “/usr/local/lib/python3.7/site-packages/homeassistant/components/velux/init.py”, line 30, in async_setup
await hass.data[DATA_VELUX].async_start()
File “/usr/local/lib/python3.7/site-packages/homeassistant/components/velux/init.py”, line 57, in async_start
await self.pyvlx.load_nodes()
File “/usr/local/lib/python3.7/site-packages/pyvlx/pyvlx.py”, line 83, in load_nodes
await self.nodes.load(node_id)
File “/usr/local/lib/python3.7/site-packages/pyvlx/nodes.py”, line 70, in load
await self._load_all_nodes()
File “/usr/local/lib/python3.7/site-packages/pyvlx/nodes.py”, line 91, in _load_all_nodes
self.add(node)
File “/usr/local/lib/python3.7/site-packages/pyvlx/nodes.py”, line 54, in add
raise TypeError()
TypeError

I'd be glad to help you out in debugging this :)

Julius2342, 2019-03-11:
This should be fixed if you run latest pyvlx (which was also merged last weekend in current dev branch of home assistant). May you retry?

gibman, 2019-03-11:
I'd like to :).

No idea how to update pyvlx on the hass.io device.
If you could give me some directions ?

Julius2342, 2019-03-11:
No idea, i just run a development environment. You could just edit the dependency within `velux/__init__.py`. Or just wait for the next release ...

gibman, 2019-03-12:
Ok- Im up and running with the latest code 0.2.10

The problem is indeed fixed as you said.

closed 2019-03-12 by gibman