Julius2342/pyvlx #18: device_class=window always ?

gibman, 2019-03-13:
Hi.

It seems like the device_class in hass.io for the "cover type" is always window for any velux product.

the cover type has several device classes to choose from. Yes I know I can override these in hass.io with the correct types.

Is it not possible to differentiate this based on what the velux api returns and then have pyvlx set this up properly ?

docs:
https://www.home-assistant.io/components/cover

----------------
snippet:

The way these sensors are displayed in the frontend can be modified in the customize section. The following device classes are supported for covers:

None: Generic cover. This is the default and doesn’t need to be set.

awning: Control of an awning, such as an exterior retractable window, door, or patio cover.

blind: Control of blinds, which are linked slats that expand or collapse to cover an opening or may be 
tilted to partially covering an opening, such as window blinds.

curtain: Control of curtains or drapes, which is often fabric hung above a window or door that can be drawn open.

damper: Control of a mechanical damper that reduces airflow, sound, or light.

door: Control of a door or gate that provides access to an area.

garage: Control of a garage door that provides access to a garage.

shade: Control of shades, which are a continuous plane of material or connected cells that expanded or 
collapsed over an opening, such as window shades.

shutter: Control of shutters, which are linked slats that swing out/in to covering an opening or may be tilted to partially cover an opening, such as indoor or exterior window shutters.

window: Control of a physical window that opens and closes or may tilt.

Julius2342, 2019-03-13:
Yes, that would be easy - and should be solved within the velux.cover component within HASS.

You should do something like:

```
if isinstance(self.node, Awning):
    return 'awning'
```

I thought on this when i implemented it, but only have windows, so this issue did not get the attention it deserves :-).

Btw do you use Siri/Homekit for controlling Hass?

gibman, 2019-03-13:
hi.

I use the the velux KLF 200 controller.

gibman, 2019-03-13:
btw. I'd be happy to test anything.

Im not too fluent with python though.

Also it seems I have to reboot the RPI each time I want to try out a new changes done to the velux.cover file.

Julius2342, 2019-03-15:
rebooting of the device sounds weirt - restarting the service should be enough - but i have no idea how to to this within hassio / hassbian. IIRC there is an option within the UI.

gibman, 2019-05-17:
I've added this to cover.py
Seems to work.

`@property
    def device_class(self):
        """Define this cover as either window, blind, awning or rullershutter."""                                     
        from pyvlx.opening_device import Blind, RollerShutter, Window, Awning                        
        if isinstance(self.node, Window):
            return 'window'        
        if isinstance(self.node, Blind):
            return 'blind'        
        if isinstance(self.node, RollerShutter):
            return 'shutter'               
        if isinstance(self.node, Awning):
            return 'awning'`

Julius2342, 2019-05-17:
Yes, that is exactly what I had in mind. May you do a PR within HASS?

Julius2342, 2019-05-17:
Please don't forget there were DualShutters added recently.

gibman, 2019-05-17:
PR added here: 
https://github.com/home-assistant/home-assistant/pull/23952

Julius2342, 2019-06-03:
Implemented with: https://github.com/home-assistant/home-assistant/pull/24279

closed 2019-06-03 by Julius2342