Julius2342/pyvlx #19: RollerShutter position w/o previous set_postion()

3s1d, 2019-04-02:
Hi,

I'm probably blind, but is there a way to get the current shutter position. I quickly tried .stop() followed by .position. However the results seem not reasonable.

Cheers,
Juergen

Julius2342, 2019-04-02:
isnt it updated automatically? That is dependent on if the house_status_monitor has send the current position already. 

(To find out, trigger a velux scene, then the house monitor sends the current positions and the nodes are updated without  a position is sent).

But: to get to the center of your question, i do not know how to trigger the busmonitor to send the positions. Iirc it does it every x minutes, but I did not try this out.

My solution is to keep pyvlx running as a daemon.

3s1d, 2019-04-02:
Hi Julius,

thanks for your quick response. 
stop() calls CurrentPosition() which updates 'position'?! (If I understand thing correctly. My Python skills are somewhat limited.) Being able to call this from the super class w/o casing a stop would really be handy. 

I read on several sites that the KLF200 has a tendency to crash if not operated in pulling mode (+ monitor disabled). The monitor will report at an interval of 15min to 24h (to preserve battery). Considering that the TCP connection will be closed after 15min idle, it may not be the the optimal solution for mains powered blinds ;)

My plan is to write a glue-program between your lib and mqtt. Similar to that one: https://github.com/3s1d/ow2mqtt . I'd like to have an update rate of 30sec or so...

Cheers,
Juergen

Julius2342, 2019-04-02:
* Yes, stop() sents the current postition (which is not the actual "current position" but a value reserved to the value "current position". )
* Pyvlx keeps the connections open by sending keep alive packets
* regarding crashes: my system here runs stable now for 4 months without any crashes...
* you should keep your daemon running and with it the connection to klf200 open - if you transmit the position to mqtt, you can just use the acutal current position saved within the internal devices of pyvlx

3s1d, 2019-04-02:
Sure thanks. I'll do so. 

The problem is if somebody uses an original remote to change the blind's position. The manipulation will take up to 15min to be detected by my system.

closed 2019-04-02 by 3s1d

Julius2342, 2019-04-04:
If pyvlx is connected, you should get the position update immediately.

3s1d, 2019-04-04:
Hi,

here's my test code:
`
    await pyvlx.load_nodes()
    await pyvlx.nodes['Buero EG Nord'].stop()
    while True:
        print(pyvlx.nodes['Buero EG Nord'].position)
        time.sleep(10)
`

I started the deamon, the position was reported correctly. Then I changed the position using a remote and waited for 20min. The position was not updated. I restarted the deamon and the correct current position was shown. .stop() seems to get the current position w/ a lag of 10-20sec (=prefect). However w/o I had no success.

Cheers,
Juergen

Julius2342, 2019-04-04:
you should use the async version of sleep ...

3s1d, 2019-04-04:
Thanks! Works a treat... Position changes arrive within 90sec, which is acceptable.

Julius2342, 2019-04-04:
uh, thats still a big delay. May you enable logging to see if you see the position changes on the packet stream...?

3s1d, 2019-04-04:
Ahm not sure how to accomplish that... 
I added "logging:...." to pyvlx.yaml; but I can't see any debug output. 

Please note:
```pyvlx/config.py:27: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  doc = yaml.load(filehandle)  
```

Julius2342, 2019-04-04:
```
"""Just a demo of the new PyVLX module."""
import asyncio

from pyvlx import Position, PyVLX

import logging
from pyvlx.log import PYVLXLOG
PYVLXLOG.setLevel(logging.DEBUG)
ch = logging.StreamHandler()
ch.setLevel(logging.DEBUG)
PYVLXLOG.addHandler(ch)

async def main(loop):
    """Demonstrate functionality of PyVLX."""
    pyvlx = PyVLX('pyvlx.yaml', loop=loop)

    # Changing position of windows:
    await pyvlx.load_nodes()

    while(True):
        for node in pyvlx.nodes:
            print(node)
        await asyncio.sleep(10)

    await pyvlx.disconnect()

if __name__ == '__main__':
    # pylint: disable=invalid-name
    LOOP = asyncio.get_event_loop()
    LOOP.run_until_complete(main(LOOP))
    LOOP.close()
```

3s1d, 2019-04-04:
Thanks. I can confirm that there is a 90sec delay between changing the position using the remote and being reported to pylvx (if I don't do stop() or something else actively).

Right after I receive this (90sec after the actual manipulation) the node is updated.
```
REC: <FrameNodeStatePositionChangedNotification node_id=0 state=5 current_position='0x0D30' target='0x0D30' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='0xF7FF' current_position_fp4='0xF7FF' remaining_time=0 time='2012-02-28 16:25:36'/>
```

Please note that I do not have any Velux products other than the KLF200. I am using Somfy blinds. The reason for using the KLF200 rather than the Somfy original (Tahoma) is that it can only be controlled using a cloud service API, which I am no fan of (due to privacy and other reasons).

The vlx2mqtt project will be available on my GitHub account and will be based upon pyvlx.

Thanks a lot!

tikismoke, 2019-04-04:
If you mad a vlx2mqtt repo I'll send you a kiss.

I still not get time to test anything until the new API is implement.

Would it allow command as well?

Know it's not the right place to ask but couldn't wait :)

phha, 2020-08-17:
It seems like this was closed without resolution. I'm still seeing this issue in Home Assistant. Is it ok to re-open this ticket?