Julius2342/pyvlx #23: Dual Roller Shutter / Expose single curtain positions

madzrobz, 2019-05-13:
The "Dual Roller Shutter" has two curtains which can be controlled separately by sending FP1 or FP2 instead of MP in GW_COMMAND_SEND_REQ.

MP: Position of the two curtains
FP1: Position of the upper curtain
FP2: Position of the lower curtain

Could this somehow be exposed to homeassistant?

Options I see:
1) create two covers for each  "Dual Roller Shutter"
2) add an additional parameter (e.g. curtain=both(default)/upper/lower) to cover.set_cover_position
3) ...?

By looking at the code, I believe it should not be too complex - but unfortunately I'm a complete newbie to python and currently lacking time to learn so I'm unable to do it myself.

Thank you!

Julius2342, 2019-05-13:
Wouldn't this be three shutters then? (top, bottom and combined). And you could ignore the ones you want.

I think we should integrate this in one pyvlx device and make it accessable via functions - on hass side we can then generate three HASS cover-devices, containing a reference to the (same) pyvlx object.

madzrobz, 2019-05-13:
You are right, to make it consistent and allow the current position to properly managed by HA, it must be 3 covers.

Arduous, 2019-06-23:
The idea of exposing 3 distinct shutters seems to be the best trade-off in terms of functionality and complexity.

Julius2342, 2019-06-23:
In a first step we should add functionality to `CommandSend` to either address "main parameter" or "functional parameter". If this works out on a low level we could can decide how to implement this on devices - and as a last step how to create devices automatically.

Julius2342, 2019-06-23:
For anyone who jumps in: the magic is here: https://github.com/Julius2342/pyvlx/blob/master/pyvlx/frames/frame_command_send.py#L31 ... :-)

madzrobz, 2021-05-07:
Pull request for addressing the single curtains created.
https://github.com/Julius2342/pyvlx/pull/66

The question remains, how to integrate with home assistant.

I still believe that simply adding a "curtain" parameter to the cover.set_position function would be sufficient and simpler then exposing three covers to home assistant (as discussed in 2019). 

I did not explore whether/how KLF 200 reports position changes on the sub-covers. I'm not sure whether we would be able to maintain the correct position of these sub-covers in home assistant. Thus only providing a one-way function call to set the position of the sub-curtains seems to be perfectly fine for me.

closed 2023-12-12 by Julius2342