Julius2342/pyvlx #24: Enable GW_COMMAND_SEND_REQ to several nodes at once from HA

madzrobz, 2019-05-13:
GW_COMMAND_SEND_REQ allows to send the command to more nodes at once.

This could be exposed to HA cover.set_cover_position by enabling more than one entity_id to be passed (however I simply don't know, whether HA allows this).

By looking at the code, I believe it should not be too complex - but unfortunately I'm a complete newbie to python and currently lacking time to learn so I'm unable to do it myself.

Thank you!

Julius2342, 2019-05-13:
Hmm, HASS handles each device separately. So it is not easy to collect position changes and sum them within one GW_COMMAND_SEND_REQ. Does this cause any problems? (latency?)

Julius2342, 2019-05-13:
> but unfortunately I'm a complete newbie

Me too :-) just started two years ago...

madzrobz, 2019-05-13:
Actually, I first ever looked at python 3 days ago, when you proposed, that I should add logging into heartbeat.py ;-) So I'm REALLY fresh here. 

I have 6 of these double roller shutters and would like to e.g. set all lower curtains (once exposed to HA) at once to a certain position. The KLF200 software seems to be kind of not robust to me, so I would prefer to just send one GW_COMMAND_SEND_REQ instead of 6 in a short period of time to not increase the risk of connection issues.

Unfortunately, specifically the lower curtains are difficult to address as it is neither possible to include them into group nor into scenes. I assume, that both groups and scenes only can set the MP and nut FP1 or FP2.

But this is optional - once we manage to expose the lower curtains to HA, I could try to position them sequentially with some delay in between.

closed 2024-09-16 by pawlizio