Julius2342/pyvlx #25: PyVLXException description="position::raw_exceed_limit" raw="b'\xd4\x00'"/

madzrobz, 2019-05-30:
The following exception was thrown:
PyVLXException description="position::raw_exceed_limit" raw="b'\xd4\x00'"/

In the logfile I only can find this:
````
2019-05-29 08:55:40 DEBUG (MainThread) [pyvlx] SEND: <FrameGetStateRequest/>
2019-05-29 08:55:40 DEBUG (MainThread) [pyvlx] REC: <FrameGetStateConfirmation gateway_state="GatewayState.GATEWAY_MODE_WITH_ACTUATORS" gateway_sub_state="GatewaySubState.IDLE"/>
2019-05-29 08:55:45 DEBUG (MainThread) [pyvlx] REC: <FrameNodeStatePositionChangedNotification node_id=6 state=2 current_position='0xC800' target='0x0000' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='0xF7FF' current_position_fp4='0xF7FF' remaining_time=0 time='1993-12-03 09:59:44'/>
2019-05-29 08:55:45 DEBUG (MainThread) [pyvlx] REC: <FrameNodeStatePositionChangedNotification node_id=6 state=77 current_position='0xC800' target='0x0000' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='0xF7FF' current_position_fp4='0xF7FF' remaining_time=0 time='1993-12-03 09:59:44'/>
2019-05-29 08:55:45 DEBUG (MainThread) [pyvlx] REC: <FrameNodeStatePositionChangedNotification node_id=6 state=5 current_position='0xC800' target='0xC800' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='0xF7FF' current_position_fp4='0xF7FF' remaining_time=0 time='1993-12-03 09:59:44'/>
2019-05-29 08:55:50 ERROR (MainThread) [homeassistant.core] Error doing job: Fatal error on SSL protocol
Traceback (most recent call last):
  File "uvloop/sslproto.pyx", line 653, in uvloop.loop.SSLProtocol._do_read
  File "uvloop/sslproto.pyx", line 728, in uvloop.loop.SSLProtocol._do_read__copied
  File "/config/deps/lib/python3.7/site-packages/pyvlx/connection.py", line 52, in data_received
    frame = frame_from_raw(raw)
  File "/config/deps/lib/python3.7/site-packages/pyvlx/frame_creation.py", line 40, in frame_from_raw
    frame.from_payload(payload)
  File "/config/deps/lib/python3.7/site-packages/pyvlx/frames/frame_node_state_position_changed_notification.py", line 49, in from_payload
    self.target = Parameter(payload[4:6])
  File "/config/deps/lib/python3.7/site-packages/pyvlx/parameter.py", line 19, in __init__
    self.raw = self.from_raw(raw)
  File "/config/deps/lib/python3.7/site-packages/pyvlx/parameter.py", line 57, in from_raw
    raise PyVLXException("position::raw_exceed_limit", raw=raw)
pyvlx.exception.PyVLXException: <PyVLXException description="position::raw_exceed_limit" raw="b'\xd4\x00'"/>
````

I don't see a suspicious package received, but maybe the exception has prevented the received data to be logged. I believe, right before the exception I wanted to set the position of node_id=6, which is a WINDOW_OPENER_WITH_RAIN_SENSOR. It was raining at that time. Maybe this is somehow connected.

Thanks for looking into this.

Julius2342, 2019-05-31:
*13 Appendix 1: Standard Parameter definition* of the API documentation says about the value `D4 00` :

*Ignore:* Ignore the parameter field where this Access Method is written

So might tell position.py to simply ignore these values.

madzrobz, 2019-06-03:
I have already implemented a similar fix - see here https://github.com/madzrobz/pyvlx/commit/2198f56d93e802bdf36348856407486791c3f407

As I understand it, the parameter value should be ignored - thus I would treat it as is the raw was None, don't even continue with the initialization and keep the position UNKNOWN.

However, since then, I was not able to reproduce the issue so formally, I cannot confirm, that it is solved.

Julius2342, 2019-06-06:
Fixed with #26

closed 2019-06-06 by Julius2342

madzrobz, 2019-06-13:
Could you please release the latest fixes as a new release?
Thank you!