Julius2342/pyvlx #27: Keep connection open to klf

ahorseman, 2019-06-25:
Dear Julius,

thank you so much for this library. It's really great to be able to control my io-homecontrol devices from the command line.

However, I'm struggling a bit with the usage, as it seems that parallel requests are not really supported and also every time I issue a command, there's a lag as the connection needs to be established.

As I am totally new to Python (I usually live in Java and Dart) I am struggling with some of the stuff and I'm not managing to get things the way I'd like to.

Is there a way to keep the connection to the device open, so the re-connect is not necessary every time I send a position to my io-homecontrol rollershutters? How would that be kept open and how can I use that? It would be great if you could extend the examples a bit.

Also is there a way to do "fire and forget" instead of the awaits?

Thanks a lot and keep this great work going.

Julius2342, 2019-06-25:
I experienced the same issue on my environment at home and I did not trace down the reason why.

pyvlx "should" keep the connection open with a heart beat mechanism. Within Home-assistant i enabled the logging for pyvlx by adding:

```
logger:
  default: warning
  logs:
    pyvlx: debug
```

I then see in my logs the following requests:

```
2019-06-25 12:17:21 DEBUG (MainThread) [pyvlx] SEND: <FrameGetStateRequest/>
2019-06-25 12:17:21 DEBUG (MainThread) [pyvlx] REC: <FrameGetStateConfirmation gateway_state="GatewayState.GATEWAY_MODE_WITH_ACTUATORS" gateway_sub_state="GatewaySubState.IDLE"/>
2019-06-25 12:18:21 DEBUG (MainThread) [pyvlx] SEND: <FrameGetStateRequest/>
2019-06-25 12:18:21 DEBUG (MainThread) [pyvlx] REC: <FrameGetStateConfirmation gateway_state="GatewayState.GATEWAY_MODE_WITH_ACTUATORS" gateway_sub_state="GatewaySubState.IDLE"/>
```

These requests should keep the connections open and reconnect if a connection died.

WRT to the fire and forget mechanism. there is such when sending a request to the API by adding `node.close(wait_for_completion=False)` to the call. But this won't skip the connection time it only does not wait until the device has reached its final position.

HTH

Julius

werner-j, 2019-06-27:
@ahorseman - I am as well rarely using Python and never digged deeper into it, so I stumbled upon similar questions. Some time ago, I set up PyVLX and have now made my efforts public and hope this helps you as well: https://github.com/werner-j/vlxcontrol

ahorseman, 2019-06-28:
Thank you. That might be of help. Though I ask myself, whether that is really clean. Perhaps @Julius2342 can use that as additional example for PyVLX and help clean it? Closing the issue. Thank you.

closed 2019-06-28 by ahorseman

Julius2342, 2019-06-28:
@werner-j : in your example, i find the name `main` pretty confusing. You should rename it to `init_pyvlx` or similar. 

I would also position the `if __name__ == '__main__':` at the very end of the file and put the `run` command within this block. 

But the most importand: Why didn't you use an async webserver framework? You should have a look at `https://aiohttp.readthedocs.io/en/stable/` ("Server example")

werner-j, 2019-06-30:
Thanks @Julius2342 - I am pretty new to Python and haven't got an overview of the existing frameworks, yet. This is a kind of learning project for me and I'm thankful for any advice.

werner-j, 2019-07-01:
Thank you for your hints @Julius2342. I've done some updates to it.