Julius2342/pyvlx #33: Tilt support for exterior blinds

pawlizio, 2020-04-04:
I tried to understand what needs to be modified to support tilt functionality for exterior blinds. It seems KLF 200 provides this with Functional Parameter 3. But the API specification and your implementation are not so easy to understand.

I moved from Somfy Tahoma to Velux KLF 200 in order to get rid of the somfy servers. However I recognized that my Raffstore don't support tilt functionality with the Velux integration.  

Would be great if someone can support.

Julius2342, 2020-04-30:
*sigh*  i broke it again. This time i will fix it ...

pawlizio, 2020-05-13:
@Julius2342 : Could you create a new release with #34 in order to adjust official HA velux component?

Julius2342, 2020-05-14:
done

closed 2020-05-14 by Julius2342