Julius2342/pyvlx #44: Incorrect Current Orientation with Somfy Venetian Blind

dumpfheimer, 2020-07-07:
Servus!

Yesterday I started migrating my Home Assistant -> Somfy Connexoon to HA -> Velux KFL 200 which worked kind of nicely, but the "tilt position" (=orientation) was missing. So I went ahead and added that to HA.

Setting the orientation works flawlessly, but it always shows strange "current orientation" values. I then set pyvlx to debug and it seems like it is receiving those strange values from the blind. But I was hoping you could confirm that for me.

_IF_ it is the blind, that is sending strange/unhandled signals, would it be possible to use the sent values as current values for this particular blind/product? It did show the right position in the Somfy App.
_IF_ it is not the blind. Do you have any idea what those values could mean? I went through the code and came to the conclusion it actually is within the "position range" which - I guess - would mean it would be a pretty inconvenient "special value"

I appreciate any help.

**Vielen Dank!**

I copied 2 debug logs from when I tried to set the orientation (it did set it correctly)

EXAMPLE 1

> Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] **SEND**: \<FrameCommandSendRequest node_ids=[3] parameter="10 %" functional_parameter="fp1: 0 %, fp2: 0 %, **fp3: 100 %**, fp4: 0 %, fp5: 0 %, fp6: 0 %, fp7: 0 %, fp8: 0 %, fp9: 0 %, fp10: 0 %, fp11: 0 %, fp12: 0 %, fp13: 0 %, fp14: 0 %, fp15: 0 %, fp16: 0 %, " session_id=56 originator=Originator.USER/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] REC: \<FrameCommandSendConfirmation session_id=56 status=CommandSendConfirmationStatus.ACCEPTED/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] REC: \<FrameNodeStatePositionChangedNotification node_id=3 state=2 current_position='0x13FC' target='0x1400' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0xF7FF**' current_position_fp4='0xF7FF' remaining_time=0 time='2032-02-08 12:54:40'/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: \<FrameNodeStatePositionChangedNotification node_id=3 state=2 current_position='0x13FC' target='0x1400' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0xF7FF**' current_position_fp4='0xF7FF' remaining_time=0 time='2032-02-08 12:54:40'/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] Living Room Left position changed to: 9 %
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] REC: \<FrameCommandRunStatusNotification session_id=56 status_id=1 index_id=3 node_parameter=0 parameter_value=5116/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] REC: \<FrameCommandRemainingTimeNotification session_id=56 index_id=3 node_parameter=0 seconds=1/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] REC: \<FrameNodeStatePositionChangedNotification node_id=3 state=4 current_position='0x13FC' target='0x1400' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0x39BD**' current_position_fp4='0xF7FF' remaining_time=1 time='2032-02-08 12:54:40'/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: \<FrameNodeStatePositionChangedNotification node_id=3 state=4 current_position='0x13FC' target='0x1400' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0x39BD**' current_position_fp4='0xF7FF' remaining_time=1 time='2032-02-08 12:54:40'/>
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] Living Room Left position changed to: 9 %
Jul  6 21:00:34 server run.sh[3602197]: 2020-07-06 21:00:34 DEBUG (MainThread) [pyvlx] **Living Room Left orientation changed to: 28 %**
Jul  6 21:00:36 server run.sh[3602197]: 2020-07-06 21:00:36 DEBUG (MainThread) [pyvlx] REC: \<FrameCommandRunStatusNotification session_id=56 status_id=1 index_id=3 node_parameter=0 parameter_value=5116/>
Jul  6 21:00:36 server run.sh[3602197]: 2020-07-06 21:00:36 DEBUG (MainThread) [pyvlx] REC: \<FrameSessionFinishedNotification session_id=56 />
Jul  6 21:00:36 server run.sh[3602197]: 2020-07-06 21:00:36 DEBUG (MainThread) [pyvlx] REC: \<FrameNodeStatePositionChangedNotification node_id=3 state=5 current_position='0x13FC' target='0x1400' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0x39BD**' current_position_fp4='0xF7FF' remaining_time=0 time='2032-02-10 01:19:12'/>
Jul  6 21:00:36 server run.sh[3602197]: 2020-07-06 21:00:36 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: \<FrameNodeStatePositionChangedNotification node_id=3 state=5 current_position='0x13FC' target='0x1400' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0x39BD**' current_position_fp4='0xF7FF' remaining_time=0 time='2032-02-10 01:19:12'/>
Jul  6 21:00:36 server run.sh[3602197]: 2020-07-06 21:00:36 DEBUG (MainThread) [pyvlx] Living Room Left position changed to: 9 %
Jul  6 21:00:36 server run.sh[3602197]: 2020-07-06 21:00:36 DEBUG (MainThread) [pyvlx] Living Room Left orientation changed to: 28 %

EXAMPLE 2

> Jul  6 21:01:09 server run.sh[3602197]: 2020-07-06 21:01:09 DEBUG (MainThread) [pyvlx] SEND: \<FrameCommandSendRequest node_ids=[3] parameter="20 %" functional_parameter="fp1: 0 %, fp2: 0 %, **fp3: 100 %**, fp4: 0 %, fp5: 0 %, fp6: 0 %, fp7: 0 %, fp8: 0 %, fp9: 0 %, fp10: 0 %, fp11: 0 %, fp12: 0 %, fp13: 0 %, fp14: 0 %, fp15: 0 %, fp16: 0 %, " session_id=57 originator=Originator.USER/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] REC: \<FrameCommandSendConfirmation session_id=57 status=CommandSendConfirmationStatus.ACCEPTED/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] REC: \<FrameNodeStatePositionChangedNotification node_id=3 state=2 current_position='0x13FC' target='0x2800' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0x39BD**' current_position_fp4='0xF7FF' remaining_time=0 time='2032-03-06 20:16:16'/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: \<FrameNodeStatePositionChangedNotification node_id=3 state=2 current_position='0x13FC' target='0x2800' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0x39BD**' current_position_fp4='0xF7FF' remaining_time=0 time='2032-03-06 20:16:16'/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] Living Room Left position changed to: 9 %
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] **Living Room Left orientation changed to: 28 %**
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] REC: \<FrameCommandRunStatusNotification session_id=57 status_id=1 index_id=3 node_parameter=0 parameter_value=5116/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] REC: \<FrameCommandRemainingTimeNotification session_id=57 index_id=3 node_parameter=0 seconds=5/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] REC: \<FrameNodeStatePositionChangedNotification node_id=3 state=4 current_position='0x13FC' target='0x2800' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0xBC3F**' current_position_fp4='0xF7FF' remaining_time=5 time='2032-03-06 20:16:16'/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: \<FrameNodeStatePositionChangedNotification node_id=3 state=4 current_position='0x13FC' target='0x2800' current_position_fp1='0xF7FF' current_position_fp2='0xF7FF' current_position_fp3='**0xBC3F**' current_position_fp4='0xF7FF' remaining_time=5 time='2032-03-06 20:16:16'/>
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] Living Room Left position changed to: 9 %
Jul  6 21:01:10 server run.sh[3602197]: 2020-07-06 21:01:10 DEBUG (MainThread) [pyvlx] **Living Room Left orientation changed to: 94 %**

dumpfheimer, 2020-07-07:
BTW I'm using pyvlx 0.2.16

bonito1, 2020-07-16:
Hello dumpfheimer,

could you explain how have you added the "tilt position" to HA exactly?
I have also venetian blinds with Somfy motors but I can not change the "tilt position" via HA. I can move them only up and down via HA.
Have you modified the the velux component?
Could you share your code to add the "tilt position" to a venetian blind?

dumpfheimer, 2020-07-16:
Hi!

You are right. HA currently does not support that. I have built it myself but have not yet tested it (especially the close tilt and open tilt correctness) and am not sure about the contribution guidelines so I did not dare to create a merge request YET.

Take a look at my changes here if you want to try it yourself:
https://github.com/dumpfheimer/core/commit/1f5006d47768e5b27ca33081794154eb6f77203c

dumpfheimer, 2020-07-20:
Hi!

I tried to investigate the values but I didn't really get smart from them. I wrote a script, that changes the orientation and reads back the response fp3 values. I put them in a CSV maybe somebody gets an idea of what could be going on.

[values_1.txt](https://github.com/Julius2342/pyvlx/files/4946760/values_1.txt)
[values_2.txt](https://github.com/Julius2342/pyvlx/files/4946761/values_2.txt)

pawlizio, 2020-08-07:
For tilt position the HA integration has to be adjusted. pyvlx already support that since #33  . For testing you can use my HACS custom component where I already modified HA integration slightly, but not yet as a final solution to create a pull request to velux integration.

I didn't modified HASS yet, as the tilt function has two different options in terms of implementation, depending on the blinds. At somfy tahoma there was the possibility to configure blinds as either as (+90°/-90°) blinds or 0-100 %. The difference is in the code. Both covers requires a value between 0 and 100 for the tilt position, but the blinds with  (+90°/-90°) are open at 50% and close at either 0% (=+90°) or 100% (=-90°) while the others are open at 0% and closed at 100% (inverted logic as in HA). In order to catch these different types of blinds, for each blind an additional configuration step is necessary to give the user the possibility to set the correct blind type. My exterior venetian blinds are from type (+90°/-90°), so I hard coded them to be open at 50 and close at 100. This is currently implemented at 

https://github.com/Julius2342/pyvlx/blob/9397bbe741623769f4137714d08538ebd29d3ff0/pyvlx/opening_device.py#L266-L281

and called from HA via: 

https://github.com/pawlizio/my_velux/blob/d52ffc7a8998f0e93aa5f5de2d99fd0c69c5d742/custom_components/my_velux/cover.py#L139-L145

However if you install custom component you can modify config\custom_components\my_velux\cover.py as follows to manually define the position_percent at which the cover is open/close 

```
    async def async_open_cover_tilt(self, **kwargs):
        """Open cover tilt."""
        orientation = Position(position_percent=0)
        await self.node.set_orientation(orientation=orientation, wait_for_completion=False )

    async def async_close_cover_tilt(self, **kwargs):
        """Close cover tilt."""
        orientation = Position(position_percent=100)
        await self.node.set_orientation(orientation=orientation, wait_for_completion=False )
```

My HACS custom component: https://github.com/pawlizio/my_velux

dumpfheimer, 2020-08-08:
Hi!

I modified the Hass code as posted above: https://github.com/dumpfheimer/core/commit/1f5006d47768e5b27ca33081794154eb6f77203c

The problem is not setting the orientation - that works fine. The problem is that it seems like the blinds are not sending the correct orientation when sending a status update. Do you have any information on that topic?

pawlizio, 2020-08-08:
You mean that you set them to 100 %, but they confirm their position at 28%?

pawlizio, 2020-08-08:
But just for clarification, physically they are at 100%? 

What is reported if you set them to another value like 25% or 30%?

dumpfheimer, 2020-08-08:
Yes, exactly. I intend to e.g. close the blinds with a 45° orientation. The blinds act exactly right and stay in the desired state. But the cakes reported back in fp3 are random other values that have nothing to do with the current orientation. Furthermore I wrote a script that set the orientation from 0-100 and displayed the responses in binary form. Ignoring the timestamp and checksum I was able to find exact the same responses from varying orientations which lets me believe the blinds do not send the orientation with the status update frame. Or it is hidden very well..

dumpfheimer, 2020-08-28:
Any suggestions what I could try?

pawlizio, 2020-08-28:
If you can't find any systematic within the feedback I only can suggest to ignore the feedback from the KLF with regards to orientation (fp3 value). In this case you would always have a latest status which you set via home assistant.

dumpfheimer, 2020-08-28:
I do have a Somfy connexoon AND the velux klf 200. The Somfy App does show the correct orientation. Should I be able to see the traffic between blind and connexoon on the klf200? Maybe I can identify a new frame.

pawlizio, 2020-08-28:
Actually all frames which are submitted by KLF200 should be visible as pyvlx debug messages.

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

dumpfheimer, 2020-08-28:
And frames sent between connexoon and the blind?

pawlizio, 2020-08-28:
No, I mean if you activate the debug mode you will see all available frames submitted by KLF. But you will not see the whole io-homecontrol communication, only the abstract which is provided by KLF200.

dumpfheimer, 2020-08-28:
Okay, too bad. Thank you, though.
I guess I will try it anyways and hope the blind might send a broadcast message or something similar that contains a hint.

dumpfheimer, 2020-08-28:
How would I enable debug mode?

pawlizio, 2020-08-28:
Enable logger for pyvlx at your configuration.yaml as follows:

```
logger:
  default: warning
  logs:
    pyvlx: debug
```

dumpfheimer, 2020-08-28:
Danke!

dumpfheimer, 2020-12-14:
Does anyone have a contact from somfy or velux whom I could ask about this?
I tried to contact them but never got a response.

dumpfheimer, 2021-03-23:
@pawlizio is my assumption correct, that you have venetian blinds too?
If so, is the second byte you receive as FP3 always 00?

I am trying to find a workaround for the random values of somfy blinds. My first try would be to ignore the value if the first byte is a position and the second byte is not 0x00

pawlizio, 2021-03-24:
@dumpfheimer yes I have exterior venetian blinds. If I set the orientation in HA the KLF always confirm only F7FF, which is defined as target valux, but pyvlx ignore this, because only values between 0 and 100% are accepted. I also don't get values on HA if I change the orientation with my remotes. But I have no random values as feedback or at least I never observed them.

Nicks57, 2021-07-23:
Hello,

I installed yesterday my new KLF200 with latest firmware as well as the Velux HA custom component from **pawlizio**. First, thanks a lot for the great contribution! 
However, I'm facing the same problems as **dumpfheimer** with my venitian blind (Warema exterior venitian blinds with Somfy IO Home control module integrated). I can perfectly control the orientation of the blades with both my remote control (Somfy Situo 5) and HA. But as current_position, I randomly get either 28% or 75%.
Is there any update on this issue? 

Thank you!

dumpfheimer, 2021-08-24:
Hi!

Nothing new on my side. I set this aside for now even though its annoying me. If more people are having this issue it might be better to ignore the FP3 from the venetian blinds all along and just cache the last set value. This of course should only happen if there are no users that actually get useful feedback on FP3, but I dont know how we could find out if such blinds exist.

There must be some kind of way to actually get the current position, because the somfy connexoon does load the right angle if you change it by eg a remote control. But I'm afraid that would be a huge trial and error mess. I wrote Velux and Somfy asking for help but neither even bothered to answer me.

Nicks57, 2021-08-24:
Hi,

In the meantime, I started to work on it. The problem comes from the frame received by the HouseMonitoring function in KLF 200 which is used for example by Home Assistant to update Main Parameter (position) and Functional Parameter (FP3 : orientation). The notification frame received periodically from HouseMonitoring contains correct values for Main Parameter but total abberant values for Functional Parameters (FP1-FP7). In our case, we get for the slats orientation values that are either UNKNOWN, 26% or 73%.... I assume this is a bug in the Velux KLF 200 API and since there is no more support, we have to find another way.
Fortunately, after some testing, there is a command that delivers the correct Functional Parameter values: GW_STATUS_REQUEST_REQ.
I implemented this command in a local copy of Pyvlx this week-end and I'm still testing it. So far, I can get the correct orientation of the slats in Python debugger but finding a solution to call the GW_STATUS_REQUEST_REQ command within the HouseMonitoring seems to be tricky.
I will continue to work on it and when a solution is found, I will submit the code for acceptance by Julius2342.

dumpfheimer, 2021-08-25:
Glad to test it if you check in your changes.

And thank you for figuring this out!

dumpfheimer, 2021-09-04:
sorry, I must have missed an email.
I will test it as soon as possible, thank you!

dumpfheimer, 2021-09-04:
Okay, I'm not home, but everything seems to be working normally and I am not getting randomly jumping values for tilt when I change the tilt position. So this looks like an improvement already. When I'm back I will test changing the orientation with the remote and see what happens.

Nicks57, 2021-09-05:
Thank you for the first feedback.
The tests with the remote were successful on my side. Orientation is updated every minute since the status is called from the heartbeat loop.

In the mean time I will finished the test files to be ready to make a pull request if the tests on your side are successful.

stephanseitz, 2021-11-16:
I've been using this modification for more than 20 days now. it still works perfectly and I couldn't find any side effects, mega! Will this PR make it into an official release? what else do you have to do?

dumpfheimer, 2021-11-17:
Hi!

I too have been using it for quite some time now without any issues.
I still have 2 concerns:

1. We should make sure the changes do not affect anybody negatively
2. There was a change in the close orientation position, which benefits the users that have -90/+90 blinds but causes my blind to close only 50% when I press "close"

First of all I dislike this because it is simply not what I need but also this is a change that might affect many other people in an unexpected undesirable way. I have spent dozens of seconds thinking about how to solve this issue but did not come up with a solution yet.

I would propose to take this change out of the PR anyways and make a separate one for that change. That would make me feel much more confident about the PR.

Nicks57, 2021-11-17:
Hi

@dumpfheimer : can you give more details on point 2? Is that a change I implemented? Can't remember....

Regarding this issue, the idea that come to my mind is to adapt the close fonction (and I assume the open fonction) to behave properly in the both cases based on a parameter value. This parameter could be set manually by the user in the configuration.yml file or automatically from the model of the actuator (readable in KLF200). I thing stephanseitz and myself have Somfy Exterior Venetian blind Type 17 and you should have another one.

dumpfheimer, 2021-11-17:
It seems like I have confused something. The change is in the master branch and has been since 2 years - unless I am blamimg the wrong line of code: https://github.com/Julius2342/pyvlx/blame/9a6d1d8988157e546e6f25bb78ccd2bfffa24f7d/pyvlx/opening_device.py#L268

Not sure why this never was an issue for me. But that definately makes things easier if I am more of an exception than the rule.
Also, this more likely affects the "open" function than "close". Hope I'm not missing anything.

BTW, are you okay with me having created the PR https://github.com/Julius2342/pyvlx/pull/109 ?
I don't want to make you feel like I am stealing your work.

Also, I would be very interested in how you found out what the issue was and how to work around it.

dumpfheimer, 2021-11-17:
Ah, one more thing. The changes in the Parameter class.
I was receiving strange values from my blind and was able to fix it by not throwing an exception but rather return UNKNOWN for those values. This is a pretty central change that should probably be reviewed by someone more knowledgeable.

I will revert that change and make a seperate PR for that, so we can move on with the blind orientation fix.

stephanseitz, 2021-11-17:
ah cool, thank you very much!

closed 2024-09-16 by pawlizio