Julius2342/pyvlx #45: Pyvlx broken when garage door is in KLF200

lmeyer, 2020-07-22:
Hello, 

I have a velux cover that works fine (KLF200 -> pyvlx -> HA).

I also have a garage door that produce an error in pyvlx when present in KLF200.

Here are the logs : 

```
2020-07-18 00:13:35 WARNING (MainThread) [pyvlx] Connected to: KLF 200: Software version: 0.2.0.0.71.0, hardware version: 6, protocol version: 3.14
ValueError: 378 is not a valid NodeTypeWithSubtype

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/asyncio/sslproto.py", line 545, in data_received
    self._app_protocol.data_received(chunk)
  File "/usr/local/lib/python3.7/site-packages/pyvlx/connection.py", line 52, in data_received
    frame = frame_from_raw(raw)
  File "/usr/local/lib/python3.7/site-packages/pyvlx/frame_creation.py", line 40, in frame_from_raw
    frame.from_payload(payload)
  File "/usr/local/lib/python3.7/site-packages/pyvlx/frames/frame_get_all_nodes_information.py", line 127, in from_payload
    self.node_type = NodeTypeWithSubtype(payload[69] * 256 + payload[70])
  File "/usr/local/lib/python3.7/enum.py", line 310, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.7/enum.py", line 564, in __new__
    raise exc
  File "/usr/local/lib/python3.7/enum.py", line 548, in __new__
    result = cls._missing_(value)
  File "/usr/local/lib/python3.7/enum.py", line 577, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))

ValueError: 378 is not a valid NodeTypeWithSubtype

2020-07-18 00:13:47 ERROR (MainThread) [homeassistant.components.velux] Can't connect to velux interface: <PyVLXException description="Unable to retrieve node information" />
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 31, in async_setup
    await hass.data[DATA_VELUX].async_start()
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 70, in async_start
    await self.pyvlx.load_nodes()
  File "/usr/local/lib/python3.7/site-packages/pyvlx/pyvlx.py", line 83, in load_nodes
    await self.nodes.load(node_id)
  File "/usr/local/lib/python3.7/site-packages/pyvlx/nodes.py", line 70, in load
    await self._load_all_nodes()
  File "/usr/local/lib/python3.7/site-packages/pyvlx/nodes.py", line 88, in _load_all_nodes
    raise PyVLXException("Unable to retrieve node information")
pyvlx.exception.PyVLXException: <PyVLXException description="Unable to retrieve node information" />
2020-07-18 00:13:47 ERROR (MainThread) [homeassistant.setup] Setup failed for velux: Integration failed to initialize.
```

378 is x17A in hex. The API doc for that type says the following:

> 0x017A Linear or angular position of the garage door

Can you add the 0x017A to the “enum” items in `NodeTypewithSubtype()` ?

Here is the discussion in HA community :
https://community.home-assistant.io/t/velux-component-for-klf-200-doesnt-support-the-new-api-with-firmware-2-0-0-71/75641/119

Many thanks !

lmeyer, 2020-07-22:
Hi @Julius2342,

I just saw your PR, you are so fast ! I modified the `const.py` file like you did "just to see".

I have now another message in logs and the garage door is not discovered in HA.
Do you have any idea ?

```
2020-07-22 19:34:22 WARNING (MainThread) [pyvlx] NodeTypeWithSubtype.LIGHT_ON_OFF not implemented
2020-07-22 19:34:22 WARNING (MainThread) [pyvlx] NodeTypeWithSubtype.LINAR_ANGULAR_POSITION_OF_GARAGE_DOOR not implemented
```

Julius2342, 2020-07-22:
The improvement is, that is no longer an exception :)

Is it possible that you had `LIGHT = 0x0180` but not `LIGHT_ON_OFF = 0x01BA` ? As far as i know `LIGHT_ON_OFF` was never implemented?

Do you know the difference? We might just treat `LIGHT_ON_OFF` the same as `LIGHT`. Then you would just have to add it to `pyvlx/node_helper.py` (you see other examples there where several subtypes are treated the same. e.g. `DUAL_ROLLER_SHUTTER`.

lmeyer, 2020-07-22:
> The improvement is, that is no longer an exception :)

For sure ! Appreciate that ;)

> Is it possible that you had `LIGHT = 0x0180` but not `LIGHT_ON_OFF = 0x01BA` ? As far as i know `LIGHT_ON_OFF` was never implemented?
> Do you know the difference? We might just treat `LIGHT_ON_OFF` [...]

I don't know how to check the hex code and I don't know the difference. I think it controls the light of the motor. I can do what you say and test.

> And the same applies to `LINAR_ANGULAR_POSITION_OF_GARAGE_DOOR` . Does this behave same or similar as a normal garage door opener?

I think it behave the same. It's a garage door (DEXXO SMART IO https://www.youtube.com/watch?v=V2VsOD4dLvo)

I will try and tell you.

closed 2020-07-31 by Julius2342

lmeyer, 2020-09-05:
Hello, 

My different tries didn't really work. As i'm not familiar with python, maybe I did things wrong.

Your fix does not seems to be included in latest HA release. 
Do you know a simple way to easily modify code and test it ? (without accessing docker container ?)

Regards,

Julius2342, 2020-09-17:
I just released a new version of pyvlx. make a pr to hass to bump the version.

PR #46 (2020-07-31): Fixed typo within enum

Fixes #45 45