Julius2342/pyvlx #47: Window opening % reported as 1 instead of 0 when closed (new vasistas)

gaggio, 2020-07-27:
Hello, I'm a proud owner of two of the newly released motorized windows with vasistas opening.
Successfully paired and operated with HA through KLF 200, but when these are completely closed the status attribute reports 1% opening instead of 0%. Thus in HA it is not reported as "closed" and the close button is not deactivated. 
Note that another window with "traditional" opening is correctly reporting 0% when closed, so it must be something related to the implementation of this new model.

Available to debug and/or provide logs.

Julius2342, 2020-07-27:
What node type id does your product have?

gaggio, 2020-07-27:
> What node type id does your product have?

I'm a bit lost on this, I can't find this information on the entity attributes nor on the logs...
I have access to the terminal and python shell though the SSH web terminal addon, is there a command/script I can run to retrieve such info?

gaggio, 2020-07-30:
Ok, enabled debug level logging, I see the windows I was talking about have: product_type=12

Julius2342, 2020-07-30:
May I ask you to paste the log?

gaggio, 2020-07-30:
Sure, here is it: https://pastebin.com/SjgNQtwW

Julius2342, 2020-07-31:
So the interesting line is:

```
2020-07-30 11:44:14 DEBUG (MainThread) [pyvlx] REC: <FrameGetAllNodesInformationNotification node_id=0 order=0 placement=0 name='Cucina' velocity=Velocity.DEFAULT node_type='NodeTypeWithSubtype.WINDOW_OPENER_WITH_RAIN_SENSOR' product_group=3 product_type=12 node_variation=NodeVariation.NOT_SET power_mode=0 build_number=28 serial_number='08:08:08:26:00:03:b5:15' state=255 
```

My Windows also identify themselves as WINDOW_OPENER_WITH_RAIN_SENSOR - and there it works correctly. 

What model are your windows? Mine are Velux roof windows from about 2017.

And if you look at the Velux KLF 200 API specs (page 114), it says:

```
Main parameter = 0x0000 100 % open
Main Parameter = 0xC800 0 % open 
```

So there is aditionally a typo / error in the API documentation.

gaggio, 2020-08-16:
These are Velux roof windows as well. These are the newly released motorized with vasistas opening. These have been added to the catalogue just last April, I ordered immediately (I was waiting for those to be released) and my authorized dealer says it's one of the first sent out and installed. I tried also with the remote, with different positions of the handle... They always go to 1 when closed.

mtdcr, 2020-09-18:
Is this a matter of calibration? I don't know whether and how it's possible to recalibrate, though. I have a cover which reports 1% only sometimes when it's closed - more often in winter than in summer.

gitmirko, 2021-06-08:
Hi, I'm having the same problem/symptom as described above. I tried to debug it, but I don't get it. I have 2 shutters and one of them is showing current_position = 1 and the state reports open.

[image]

Here are some debug logs, but I can't spot the difference that results in the position being 1:
```
2021-06-08 21:48:56 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: <FrameGetAllNodesInformationNotification node_id="3" order="3" placement="0" name="Velux Rollo Treppe" velocity="Velocity.DEFAULT" node_type="NodeTypeWithSubtype.ROLLER_SHUTTER" product_group="1" product_type="1" node_variation="NodeVariation.NOT_SET" power_mode="0" build_number="16" serial_number="XXX" state="255" current_position="0xF7FF" target="0xF7FF" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2021-06-08 21:48:49" alias_array=""/>
2021-06-08 21:48:56 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: <FrameGetAllNodesInformationNotification node_id="0" order="0" placement="0" name="Velux Rollo Abstellraum" velocity="Velocity.DEFAULT" node_type="NodeTypeWithSubtype.ROLLER_SHUTTER" product_group="1" product_type="1" node_variation="NodeVariation.NOT_SET" power_mode="0" build_number="16" serial_number="XXX" state="255" current_position="0xF7FF" target="0xF7FF" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2021-06-08 21:48:49" alias_array=""/>
2021-06-08 21:48:56 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: <FrameGetAllNodesInformationNotification node_id="1" order="1" placement="0" name="Velux Abstellraum" velocity="Velocity.DEFAULT" node_type="NodeTypeWithSubtype.WINDOW_OPENER_WITH_RAIN_SENSOR" product_group="3" product_type="1" node_variation="NodeVariation.NOT_SET" power_mode="0" build_number="48" serial_number="XXX" state="255" current_position="0xF7FF" target="0xF7FF" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2021-06-08 21:48:49" alias_array="d802=6400, d803=ba00"/>
2021-06-08 21:48:56 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: <FrameGetAllNodesInformationNotification node_id="2" order="2" placement="0" name="Velux Treppe" velocity="Velocity.DEFAULT" node_type="NodeTypeWithSubtype.WINDOW_OPENER_WITH_RAIN_SENSOR" product_group="3" product_type="1" node_variation="NodeVariation.NOT_SET" power_mode="0" build_number="48" serial_number="XXX" state="255" current_position="0xF7FF" target="0xF7FF" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2021-06-08 21:48:49" alias_array="d802=6400, d803=ba00"/>
```
Best regards
Mirko

Nicks57, 2021-08-01:
Hi,

The problem here is coming from the inaccuracy of some motors. When you set the position of the cover to 51200 (this value correspond to 100% close in the KLF200 API. Range in the API is 0 to 51200 which represent C800 in Hexadecimal), the cover will close and send back the position. HA reads then this position and convert it in integer and percentage (100 - the value read). However, as you are expecting that the cover will send back 51200 when it is closed, some cover will sent a bit less. For Velux SML covers for example, I read 51160 (I use the API to directly read what is coming out of the KLF200). Some over cover seams to be more accurate. My SOMFY Blinds respond exactly 51200 when closed.

So when we get 51160, the conversion done in PYVLX results in 99% and not 100% (100% will come out only if you have exactly 51200). That’s why we get 1% (100-99) instead of 0.

A simple solution will be to slightly modify the conversion done in PYVLX, which takes place in the parameter.py file, function « to_percent(raw) ». For now the formula is int(raw[0] / 2). I suggest to change it to int((raw[0] / 2) + 0.5). Adding the 0.5 will solve the problem of inaccuracy that we have with Velux cover.
I tested it out at home and I get the correct value. The Somfy blinds positions are also correct with this modification.

I hope that Julius2342 will agree to this and make the modifications.

Thank you!

Julius2342, 2021-08-02:
@Nicks57 : I think your considerations sound reasonable. May I ask you to do a PR? (It would be great if you could add a comment to the line of code, describing the reason behind the formula).

Thank you in advance!

Julius

Nicks57, 2021-08-02:
@Julius2342 Sorry but I'm not very familiar to GitHub: to which branch should I do the Pull Request?

Julius2342, 2021-08-02:
To `master` . I can help if you encounter problems ...

Nicks57, 2021-08-02:
@Julius2342 I would appreciate your help. Create Pull Request to Master is greyed out here...

Thx
Nicolas

Julius2342, 2021-08-02:
Is it to master on this repo?

Julius2342, 2021-08-02:
you have to push your changes first to a remote branch (!= master) on your repo ( git@github.com:Nicks57/pyvlx.git ).

If you have a local checkout:

```
git remote add nick git@github.com:Nicks57/pyvlx.git
git checkout -b somebranchname
git push nick somebranchname
```

Nicks57, 2021-08-02:
@Julius2342 Done! Hope I made no mistake. 

Thank you for your help! Man lernt nie aus :-)
Nicolas

Julius2342, 2021-08-03:
Your PR goes into master of your repository. You have to change the target branch to `Julius2342/pyvlx`

PR #72 (2021-08-07): Improvement of percent calculation
Improvement of percent calculation in to_percent() for devices that do not return exactly 51200 as final position when closed.

Fixes #47
closed 2021-08-07 by Julius2342