Julius2342/pyvlx #54: PYPI release in 0.2.17 is not really 0.2.16

sbyx, 2020-11-02:
It seems that when I'm doing a pip3 install pyvlx I'm not getting the 0.2.17 release of pyvlx even though it tells me it is 0.2.17
I noticed e.g. that in the connection.py of pyvlx still has this line (which is causing me trouble):

```
ssl_handshake_timeout=5,
```

even though it is not there in the release commit for 0.2.17: https://github.com/Julius2342/pyvlx/blob/9397bbe741623769f4137714d08538ebd29d3ff0/pyvlx/connection.py

I guess this might be related to #53 and maybe also #52.
Home Assistant is affected as well by this.

Julius2342, 2020-11-02:
Bumped a new release. May I ask you to try out 0.2.18 ?

sbyx, 2020-11-02:
I can confirm at least the offending line with the ssl_handshake_timeout is gone now after upgrading to 0.2.18 release using pip. If you want me to do any more tests please let me know.

sbyx, 2020-11-02:
Thanks a lot for the quick fix!

closed 2020-11-02 by sbyx