Julius2342/pyvlx #57: Velux component: asynchronous startup

dumpfheimer, 2020-12-14:
Hi!

I know this is actually in the velux component in hass, but I just saw a similar issue and thought I might address it here first (also because I do not have time to dig in to hass contribution stuff):

I believe the velux component may not be connecting at the optimal stage according to hass. (this is a wild accusation I came up with when looking at the hue integration)
I tried to move things around a bit and came up with this:

https://github.com/dumpfheimer/core/commit/9308fea1c90ae78c8bfa946b31ae20475b04ce0e#diff-78e93eab257b26b0e223ef534c57c4cf98768ef63ffab7da79a9d8ea2e9f915c

The advantage of this approach was:
- The startup of hass is not delayed
- The entities are set up after pyvlx was able to establish a connection. If it fails at first, no restart is needed.

Things I am not quite sure about yet:
- From reading the hass code I would believe that hass retries to load the component with increasing timeouts (limited at a max of 5 minutes?!). But it happened once or twice that hass gave up loading the component. Not sure why

I would appreciate feedback and or critics of someone with more experience concerning hass and pyvlx.

Julius2342, 2020-12-14:
Hi @dumpfheimer ,

when i implemented the component, the whole hass-setup process was not yet as sophisticated as it is now.

And yes, it would be good to move the connection out of the boot process itself, so it does not block the start. But i have to few insights on the current platform stucture of hass, to see if your approach works.

Did you see: https://github.com/home-assistant/core/pull/42773

dumpfheimer, 2020-12-14:
I have not, and this looks much more sophisticated than what I was up to.
I see he has also added the tilt functionality which would have been my next question.

Guess I should have waited a bit longer ;-)

dumpfheimer, 2020-12-14:
Oh, and thanks for pointing it out and answering so quickly

closed 2020-12-14 by dumpfheimer

Julius2342, 2020-12-14:
:-)

(but that PR is now idling for a while, i guess it would be great if you could review it and try it out)