Julius2342/pyvlx #58: DEXXO SMART io not working

otokarl, 2020-12-29:
Hi,

My garage door DEXXO SMART io from Somfy is not recognized.
I already have a minor fix to make it recognized by pyvlx.
Please add me as a contibutor so that I could submit a pull request.

Regards

[https://www.somfy.fr/produits/1216540/dexxo-smart-io-pack-confort](https://www.somfy.fr/produits/1216540/dexxo-smart-io-pack-confort)

Julius2342, 2020-12-29:
Hi @otokarl ,

thank you for your contribution!

just clone the repo to your account - push to this remote and do a pull request.

(that is the common approach / workflow)

Best

Julius

otokarl, 2020-12-30:
Hi,
When I try to push my local branch to github, I get this error.
I've missed something?

git --no-optional-locks -c color.branch=false -c color.diff=false -c color.status=false -c diff.mnemonicprefix=false -c core.quotepath=false -c credential.helper=sourcetree push -v --set-upstream origin refs/heads/otokarl_dexxo:refs/heads/otokarl_dexxo 
Pushing to https://github.com/Julius2342/pyvlx.git
**remote: Permission to Julius2342/pyvlx.git denied to otokarl.
fatal: unable to access 'https://github.com/Julius2342/pyvlx.git/': The requested URL returned error: 403**
Completed with errors, see above

Julius2342, 2020-12-30:
You have to clone the repository into your account ( https://github.com/otokarl/pyvlx ).

Then you can add this repository as remote: `git remote add someusefulname git@github.com:otokarl/pyvlx.git`

Then you have to remotes (verify with `git remote -v`, you should see `origin` and `someusefulname`).

Then you can do `git push -u someusefulname nameofyourbranch`

otokarl, 2020-12-30:
Ok thanks,
Now the build failed but I don't know why, and I don't see the link with my changes.
Could you please take a look ?
[https://github.com/Julius2342/pyvlx/pull/59](https://github.com/Julius2342/pyvlx/pull/59)

closed 2020-12-30 by Julius2342

otokarl, 2020-12-30:
Thanks @Julius2342 !
When do you plan a new release ?