Julius2342/pyvlx #60: [Feature request] Support silent mode

mathieu-mp, 2021-01-06:
Hi,

Thank you for your job on klf200 !

Would you consider supporting silent mode ?
Silent mode is a slow speed motor mode on Somfy "S&SO RS100 io" motors. "Smoove RS100 io" remote has a button to enable this feature, but I cannot tell if klf200 supports it.

Regards, Mathieu.

Julius2342, 2021-01-12:
Happy to accept pull requests :-)

mathieu-mp, 2021-03-11:
The klf200 API doc mentions a removal of the motor speed endpoint
We won't go any further on this feature with the current firmware

closed 2021-03-11 by mathieu-mp

pawlizio, 2021-03-11:
Isn't it possible to define the motor speed with a fictional parameter. I saw such a solution on FHEM forum and source code. I could not observe any change in speed on my blinds, but actually I'm not sure if my support this feature. I have Somfy blinds and covers and with Tahoma I also don't have this feature.

Julius2342, 2021-03-13:
I remember that I tried that out, but it did not work. So i did not add it to the interface...