Julius2342/pyvlx #61: Build failure with Python 3.10

fabaff, 2021-03-03:
If running with Python 3.10 then the tests don't pass.

```bash
[...]
self.name)
                        )
E               AttributeError: Command: no attribute
'GW_LEAVE_LEARN_STATE_REQ'

/usr/lib64/python3.10/enum.py:146: AttributeError
=========================== short test summary info
============================
FAILED
test/frame_leave_learn_state_req_test.py::TestFrameLeaveLearnStateRequest::test_frame_from_raw
======================== 1 failed, 287 passed in 0.84s
=========================
```

See https://bugzilla.redhat.com/show_bug.cgi?id=1928062 for details.

Tanks

Julius2342, 2021-03-03:
3.10 ... cutting edge :)

tirkarthi, 2021-03-07:
This could be probably due to https://bugs.python.org/issue43162 . Looking at the file everything seems to access from enum while this if clause alone retrieves from the object. I guess it's probably a typo of capital C and lowercase C.

https://github.com/Julius2342/pyvlx/blob/2c19f35ca2f5cb9d318a0e11b8726e388738f8d6/pyvlx/api/frame_creation.py#L180

PR #62 (2021-03-08): Access enum from the Enum class instead of instance which is deprecated since Python 3.10.
Fixes #61
closed 2021-03-08 by Julius2342