Julius2342/pyvlx #68: pyvlx ssl handshake error 2021.6.0 & 2021.6.1

g3r0nim00, 2021-06-04:
**The problem**

When I perform the home-assistant upgrade to 2021.6.0 & 2021.6.1, the velux integration no longer works.
Logging shows it terminates with the famous SSL handshake message.
Rebooting the KLF, which mostly solves issues after HA upgrade/reboot/..., doesn't work. Once the KLF reboot is initiated I also reloaded HA to restart the integration loading.
Downgrading to core-2021.5.5 solves the issue, the integration loads without errors.

**What is version of Home Assistant Core has the issue?**

core-2021.6.0 & core-2021.6.1

**What was the last working version of Home Assistant Core?**

core-2021.5.5

**Anything in the logs that might be useful for us?**

Version supervisor-2021.05.4
Operating System Home Assistant OS 5.13
Deployment production

Logger: homeassistant.setup
Source: components/velux/__init__.py:76 
First occurred: 8:51:57 (1 occurrences) 
Last logged: 8:51:57

Error during setup of component velux
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 248, in _async_setup_component
    result = await task
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 31, in async_setup
    await hass.data[DATA_VELUX].async_start()
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 76, in async_start
    await self.pyvlx.load_scenes()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 80, in load_scenes
    await self.scenes.load()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/scenes.py", line 51, in load
    await get_scene_list.do_api_call()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/api/api_event.py", line 21, in do_api_call
    await self.send_frame()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/api/api_event.py", line 33, in send_frame
    await self.pyvlx.send_frame(self.request_frame())
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 66, in send_frame
    await self.connect()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 43, in connect
    await self.connection.connect()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/connection.py", line 87, in connect
    self.transport, _ = await self.loop.create_connection(
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1050, in create_connection
    transport, protocol = await self._create_connection_transport(
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1080, in _create_connection_transport
    await waiter
ConnectionAbortedError: SSL handshake is taking longer than 60.0 seconds: aborting the connection

g3r0nim00, 2021-06-07:
same for core-2021.6.2
a rollback to 2021.5.5 solves the issue

g3r0nim00, 2021-06-08:
same for core-2021.6.3

`Logger: homeassistant.setup
Source: components/velux/__init__.py:76 
First occurred: 9:36:28 (1 occurrences) 
Last logged: 9:36:28

Error during setup of component velux
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 248, in _async_setup_component
    result = await task
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 31, in async_setup
    await hass.data[DATA_VELUX].async_start()
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 76, in async_start
    await self.pyvlx.load_scenes()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 80, in load_scenes
    await self.scenes.load()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/scenes.py", line 51, in load
    await get_scene_list.do_api_call()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/api/api_event.py", line 21, in do_api_call
    await self.send_frame()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/api/api_event.py", line 33, in send_frame
    await self.pyvlx.send_frame(self.request_frame())
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 66, in send_frame
    await self.connect()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 43, in connect
    await self.connection.connect()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/connection.py", line 87, in connect
    self.transport, _ = await self.loop.create_connection(
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1050, in create_connection
    transport, protocol = await self._create_connection_transport(
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1080, in _create_connection_transport
    await waiter
ConnectionAbortedError: SSL handshake is taking longer than 60.0 seconds: aborting the connection

`

g3r0nim00, 2021-06-24:
@Julius2342 this still remains an issue for me everytime I upgrade/restart/... 
I can only say it started sinds one of the core-2016.6.x upgrades.
Aren't you experiencing the same issues? I assume you also use the klf-200 with home-assistant?

vaandefanel, 2021-06-25:
same on 2021.6.5

`
Logger: homeassistant.setup
Source: components/velux/__init__.py:76
First occurred: 01:19:25 (1 occurrences)
Last logged: 01:19:25

Error during setup of component velux
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 248, in _async_setup_component
    result = await task
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 31, in async_setup
    await hass.data[DATA_VELUX].async_start()
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 76, in async_start
    await self.pyvlx.load_scenes()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 80, in load_scenes
    await self.scenes.load()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/scenes.py", line 51, in load
    await get_scene_list.do_api_call()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/api/api_event.py", line 21, in do_api_call
    await self.send_frame()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/api/api_event.py", line 33, in send_frame
    await self.pyvlx.send_frame(self.request_frame())
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 66, in send_frame
    await self.connect()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/pyvlx.py", line 43, in connect
    await self.connection.connect()
  File "/usr/local/lib/python3.8/site-packages/pyvlx/connection.py", line 87, in connect
    self.transport, _ = await self.loop.create_connection(
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1050, in create_connection
    transport, protocol = await self._create_connection_transport(
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1080, in _create_connection_transport
    await waiter
ConnectionAbortedError: SSL handshake is taking longer than 60.0 seconds: aborting the connection
`

g3r0nim00, 2021-06-28:
@vaandefanel : for me it mostly kicks in after a 'couple' restart cores... but it's far from ideal because 'couple' is undefined and sometimes it takes like 20 restarts before it finally works again.
I have it running now in core-2021.6.6

dumpfheimer, 2021-06-29:
Hi! Swooping in here randomly:
Are you positive it actually has to do with the HA core version?
The timeouts can randomly occur when the gateway has both connections occupied (KLF200 can only handle 2 connections, they time out after 15 minutes).

Maybe try upgrading hass, unplug KLF a few seconds, start KLF, start HA and see if it connects.

g3r0nim00, 2021-06-29:
Hi,

Each time I did the upgrade I did restart the KLF at a given point...
When rolling back, the issue was resolved... 
I don't want to pin-point it to the core version, I'm wondering if other users have core-2021.6.x and the KLF without issues

dumpfheimer, 2021-06-29:
I have no issues using:
HA 2021.6.6
With this extra PR: https://github.com/home-assistant/core/pull/42773
pyvlx installed from git

I would first try updating pyvlx and if that does not fix it try the PR

tbozo, 2022-07-07:
Hi,
It was working almost fine for me for some time but now even after multiple reboot I have :
```Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 235, in _async_setup_component
    result = await task
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 39, in async_setup
    await hass.data[DATA_VELUX].async_start()
  File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 84, in async_start
    await self.pyvlx.load_scenes()
  File "/usr/local/lib/python3.9/site-packages/pyvlx/pyvlx.py", line 80, in load_scenes
    await self.scenes.load()
  File "/usr/local/lib/python3.9/site-packages/pyvlx/scenes.py", line 51, in load
    await get_scene_list.do_api_call()
  File "/usr/local/lib/python3.9/site-packages/pyvlx/api/api_event.py", line 21, in do_api_call
    await self.send_frame()
  File "/usr/local/lib/python3.9/site-packages/pyvlx/api/api_event.py", line 33, in send_frame
    await self.pyvlx.send_frame(self.request_frame())
  File "/usr/local/lib/python3.9/site-packages/pyvlx/pyvlx.py", line 66, in send_frame
    await self.connect()
  File "/usr/local/lib/python3.9/site-packages/pyvlx/pyvlx.py", line 43, in connect
    await self.connection.connect()
  File "/usr/local/lib/python3.9/site-packages/pyvlx/connection.py", line 87, in connect
    self.transport, _ = await self.loop.create_connection(
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1090, in create_connection
    transport, protocol = await self._create_connection_transport(
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1120, in _create_connection_transport
    await waiter
ConnectionAbortedError: SSL handshake is taking longer than 60.0 seconds: aborting the connection
```
My config:
Home Assistant Core 2022.6.7
Home Assistant Supervisor 2022.06.2
Home Assistant OS 8.2

For what I understand with stock we're still one version behind pyvlx latest.
Do you have a "magic" reboot sequence or some hack to make it work ?

closed 2024-09-16 by pawlizio