Julius2342/pyvlx #99: Inverted status for awnings

WhistleMaster, 2021-10-03:
Hi !

Thanks for the great work on that project ! 

I have an issue awnings, the status is inverted. When they are closed, the status is "open" when they are closed and vice-versa.

Kind regards,
WM

entmoet, 2022-04-19:
Yes, me too. The postion is inverted

T-E-N, 2022-09-09:
The logic somewhat counterintuitively for awnings (but understandably due to similar motors) is as for roller shutters (with Somfy at least): close means max. shade; open letting all sunshine in.

T-E-N, 2026-02-21:
It's only been a matter of terminology: Contrary to a vertical roller blind, colloquially most people would consider a sunshade "open" when "all rolled out" and closed (its arms folded) when "all rolled up" - then again making this distinction would call the same motor position differently depending on what it's attached to, i.e. just shift the inconsistency elsewhere.