Julius2342/pyvlx #123: Different venetian blinds are 'open' at different orientation values

dumpfheimer, 2021-11-23:
There seem to be two types of venetian blinds, one goes from -90° to + 90° (called type A from now on) another goes from 0° to 90° (called type B from now on).
The problem emerges on the "open" command. At the moment, when you 'open' the orientation, the 50% orientation is sent. For type A blinds this is the correct behavior, but type B blinds will be at 45°.

The topic seems to get more and more complicated as you get into it.
After my short research I believe there is no representation in HA of a -90/+90 blind, so the interface to HA will probably be the same for both (vs having -100% (closed) to 0% (open) to +100% (closed) in HA).

This takes away the more complicated solutions anyway. Pretty much the only thing left is changing the "open" value, as proposed by @pawlizio in [Link 2](<https://github.com/Julius2342/pyvlx/issues/44#issuecomment-670699531>).
So the "only thing left to do" is to distinguish between the two types.

There can/should be two ways to do that: automatically and manual.

1. Automatic: would depend on a unique feature of the types, as mentioned in [Link 3](https://github.com/Julius2342/pyvlx/issues/44#issuecomment-971388244). My type B blind is EXTERIOR_VENETIAN_BLIND 0x0440.
2. Manual: configure in HA configuration.yaml and pass the information to pyvlx somehow
3. Manual: configure in HA Customizations (not sure if this is a UX benefit) and pass the information to pyvlx somehow

I, of course, would prefer the automatic solution, but am not familiar enough with the IO Homecontrol protocol / definitions to know if the distinction is possible with enough certainty to rely on it (Tahoma seems to let the user choose, which makes me less confident, see [Link 4](https://github.com/Julius2342/pyvlx/issues/44#issuecomment-951605853))

Any thoughts and information on this topic are/is very appreciated.

Links:
1. Discussion on "open" vs "close" percent with a comment about 90/90 blinds:
https://community.home-assistant.io/t/why-the-heck-cant-we-freely-configure-a-cover-to-interpret-its-percentage-value-as-percentage-open-or-percentage-closed/225453/64
2. Comment on 90/90 blind issue:
https://github.com/Julius2342/pyvlx/issues/44#issuecomment-670699531
3. Comment about Type identification
https://github.com/Julius2342/pyvlx/issues/44#issuecomment-971388244
4. Comment and screenshot about the Tahoma solution (manual)
https://github.com/Julius2342/pyvlx/issues/44#issuecomment-951605853

pawlizio, 2021-11-25:
I've prepared and tested a configuration posibility via HA. For open_orientation and close_orientation I've created NumberEntities which can be adjusted by a slider in HA between a range of 0 and 100. So you can define your desired open and close orientation which will be used if you press the arrow buttons. You can try this also using my custom compent, but you have to use the branch "add_orientation_config": https://github.com/pawlizio/my_velux/tree/add_orientation_config

dumpfheimer, 2021-12-14:
Oh man, I totally missed this, this is really cool, thank you!
I am currently still using your removed pull request from a few months ago which added config flow capabilities.
Am I at risk of messing things up when I also install the custom component?

Thank you! And sorry for my late response

closed 2023-12-12 by Julius2342