Julius2342/pyvlx #145: Get position while moving a cover

tschevko, 2022-01-21:
Dear all,

I'm currently testing the KLF200 (Firmware 2.0.0.71) to control my 4 solar covers via Home Assisstant. I see in the Home Assistant Log (activated for Velux via configuration.yaml), that the current position is only sent when the cover stops.

I would like to implement a kind of protection, so that when the window is open and someone want to close it via the  normal remote control (KLI 313 WW), the cover shall stop. Unfortunately I can't detect weather the cover is moving without an updated position during movement. Also it would make the visual representation of the cover closing in Home Assistant more "smooth".

As for the stop command the current position can be requested I would assume this should be possible to implement or maybe it is already implemented and I missed it.

Is there a way to get the position while the cover is moving (also in the "current_position" attribute for Home Assistant)?

Br

Julius2342, 2022-01-25:
Hmm, my velux windows report the position during movement ...

redneckdba, 2022-01-25:
After moving from a Somfy Connexoon box to the KLF-200 I have noticed this as well for all my Velux SSL Shutters and Somfy Motors as I'm using the current position / state in various automations within HomeAssistant and they simply won't work any longer due to this. As the reporter mentioned, as soon as the motor is stopped, the current position in HA gets updated immediately.

tschevko, 2022-01-26:
@Julius2342 First of all, thanks for your feedback! I'm not a coder, but I tried to read the code and I thought, in terms of position aquisition, windows and covers are handled in the same way. Would it be possible to force a higher (maybe configurable) position update rate for covers? I mean if you press the stop button also the current position can be gathered immediately.

I tried to find an "easy" solution in the code, but I could not find it.

@redneckdba Thank you for the confirmation, so it seems it is not a specific behaviour of my setup and there are use cases for which a more frequently updated position is needed.

T-E-N, 2022-09-09:
Using just a .py without HomeAssistant, I can confirm the following gets the Awning and RollerShutter names nicely (incl. spaces and diacritics), performs set_position to percentages as well as open or close, but in either case comes up empty for serial_number and position on all 7 of our Somfy motors/actuators:
```
    await pyvlx.load_nodes()
    print(*pyvlx.nodes)
<... node_id="0" serial_number="None" position="UNKNOWN"/>
```
Is there a method to populate/query these, position in particular?

closed 2023-12-12 by Julius2342