Julius2342/pyvlx #153: Integration failed to initialize

andrazek, 2022-02-11:
Hello. I really appreciate your work!!

But I'm asking for a little help. I want to configure the automation of my pergola (with Somfy io motors/controllers). Sadly, I realized one of them breaks the Velux integration. After adding several vertical awnings to KLF200, everything worked fine, but when I added Somfy LED dimmer and horizontal awning (pergola blades) & rebooted the gateway & HA the integration failed.

I suspect horizontal awning is causing troubles as node 1025 (0x0401) is fairly close to the 0x0400 mentioned in the API to be the horizontal awning. It is also displayed under "My products" in the Velux KLF200 web interface as the horizontal awning. I really need this one to work (and probably many others will follow). Any suggestion, solution?

Log output:
```
2022-02-11 00:18:04 ERROR (MainThread) [homeassistant] Error doing job: application protocol failed to receive SSL data
Traceback (most recent call last):
File "/usr/local/lib/python3.9/asyncio/sslproto.py", line 545, in data_received
self._app_protocol.data_received(chunk)
File "/usr/local/lib/python3.9/site-packages/pyvlx/connection.py", line 53, in data_received
frame = frame_from_raw(raw)
File "/usr/local/lib/python3.9/site-packages/pyvlx/api/frame_creation.py", line 50, in frame_from_raw
frame.from_payload(payload)
File "/usr/local/lib/python3.9/site-packages/pyvlx/api/frames/frame_get_all_nodes_information.py", line 144, in from_payload
self.node_type = NodeTypeWithSubtype(payload[69] * 256 + payload[70])
File "/usr/local/lib/python3.9/enum.py", line 384, in __call__
return cls.__new__(cls, value)
File "/usr/local/lib/python3.9/enum.py", line 702, in __new__
raise ve_exc
ValueError: 1025 is not a valid NodeTypeWithSubtype
2022-02-11 00:18:05 WARNING (MainThread) [homeassistant.setup] Setup of velux is taking over 10 seconds.
2022-02-11 00:18:13 ERROR (MainThread) [homeassistant.components.velux] Can't connect to velux interface: <PyVLXException description="Unable to retrieve node information" />
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 39, in async_setup
await hass.data[DATA_VELUX].async_start()
File "/usr/src/homeassistant/homeassistant/components/velux/__init__.py", line 85, in async_start
await self.pyvlx.load_nodes()
File "/usr/local/lib/python3.9/site-packages/pyvlx/pyvlx.py", line 76, in load_nodes
await self.nodes.load(node_id)
File "/usr/local/lib/python3.9/site-packages/pyvlx/nodes.py", line 69, in load
await self._load_all_nodes()
File "/usr/local/lib/python3.9/site-packages/pyvlx/nodes.py", line 87, in _load_all_nodes
raise PyVLXException("Unable to retrieve node information")
pyvlx.exception.PyVLXException: <PyVLXException description="Unable to retrieve node information" />
2022-02-11 00:18:13 ERROR (MainThread) [homeassistant.setup] Setup failed for velux: Integration failed to initialize.
```

andrazek, 2022-02-12:
Hello. I changed the value of HORIZONTAL_AWNING in const.py from 0x0400 to 0x0401, just to try and it worked. But the current position is messed up. It simply shows a negative value and it doesn't change when opening/closing. Can this be related to the wrong node type with subtype? I mean, should I investigate the pyvlx code or it's more likely a problem with the device (maybe incorrect set up)?

andrazek, 2022-02-15:
The "target position" seems to work nicely; it rotates the blades to the desired position, but the current position is never updated (always 0xF7FF). Could it be, that some other parameter contains the position? It's not evident from the debug logs. Which parameter is for the tilt position (e.g., for the venetian blinds). Please help!

I have a [Somfy slim receiver IO plug](https://www.somfy.de/produkte/1811584/pergola-slim-receiver-io-plug). I also tried to readjust the end positions and it was the same (target pos. worked, current pos. didn't work).

```
2022-02-15 07:15:22 DEBUG (MainThread) [pyvlx] SEND: <FrameCommandSendRequest node_ids="[6]" parameter="8 %" functional_parameter="fp1: 0 %, fp2: 0 %, fp3: 0 %, fp4: 0 %, fp5: 0 %, fp6: 0 %, fp7: 0 %, fp8: 0 %, fp9: 0 %, fp10: 0 %, fp11: 0 %, fp12: 0 %, fp13: 0 %, fp14: 0 %, fp15: 0 %, fp16: 0 %, " session_id="31" originator="Originator.USER"/>
2022-02-15 07:15:22 DEBUG (MainThread) [pyvlx] REC: <FrameCommandSendConfirmation session_id="31" status="CommandSendConfirmationStatus.ACCEPTED"/>
2022-02-15 07:15:23 DEBUG (MainThread) [pyvlx] REC: <FrameNodeStatePositionChangedNotification node_id="6" state="2" current_position="0xF7FF" target="0x1000" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2006-08-29 10:51:12"/>
2022-02-15 07:15:23 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: <FrameNodeStatePositionChangedNotification node_id="6" state="2" current_position="0xF7FF" target="0x1000" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2006-08-29 10:51:12"/>
2022-02-15 07:15:23 DEBUG (MainThread) [pyvlx] REC: <FrameCommandRunStatusNotification session_id="31" status_id="1" index_id="6" node_parameter="0" parameter_value="63487"/>
2022-02-15 07:15:23 DEBUG (MainThread) [pyvlx] REC: <FrameCommandRemainingTimeNotification session_id="31" index_id="6" node_parameter="0" seconds="0"/>
2022-02-15 07:15:24 DEBUG (MainThread) [pyvlx] REC: <FrameCommandRunStatusNotification session_id="31" status_id="1" index_id="6" node_parameter="0" parameter_value="63487"/>
2022-02-15 07:15:24 DEBUG (MainThread) [pyvlx] REC: <FrameSessionFinishedNotification session_id="31"/>
2022-02-15 07:15:24 DEBUG (MainThread) [pyvlx] REC: <FrameNodeStatePositionChangedNotification node_id="6" state="5" current_position="0xF7FF" target="0x1000" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2006-08-30 05:03:28"/>
2022-02-15 07:15:24 DEBUG (MainThread) [pyvlx] NodeUpdater process frame: <FrameNodeStatePositionChangedNotification node_id="6" state="5" current_position="0xF7FF" target="0x1000" current_position_fp1="0xF7FF" current_position_fp2="0xF7FF" current_position_fp3="0xF7FF" current_position_fp4="0xF7FF" remaining_time="0" time="2006-08-30 05:03:28"/>
```

andrazek, 2022-02-16:
Another update, maybe someone finds it useful.

This time I set in const.py `LOUVER_BLIND= 0x0401`. Now, after sending the target position the current position in HA is set correctly. But this is probably a software solution... from the logs it's still evident that the value is not received from the KLF200. So in case, I press stop/open/close the current position falls back to a negative value (so the information is lost). It is also not updated if the position is altered by the remote. Is this some kind of bug of the KLF device? Is something wrong with the Somfy IO receiver plug? I hope for the feedback of someone with the same type of those Somfy slim receiver IO plugs...

nick-mitigo, 2023-03-27:
I'm facing also this issue. Attached the log:
[log.txt](https://github.com/Julius2342/pyvlx/files/11075301/log.txt)

KLF-200 is managing 4 SOmfy motors, probably via such a device: https://www.somfy.co.uk/products/1811131/evb-slim-receiver-for-variation-io-plug

Any update on this issue?

alucryd, 2023-08-23:
I believe I'm affected by this as well. Got 12 somfy roller shutters paired, and one gate. For some reason 2 of the shutters are stuck at -24% in HA, I can open and close them just fine, but their position is never updated. I would say that's an issue maybe with the klf200, or more likely the shutters themselves as I've got 10 of them working fine.

Here are working and non working position from the pyvlx api, you can see that the non-working one is at 54272 and reports being neither closed nor open.

[image]
[image]

I tried repairing them to the klf200 but it didn't help. Any ideas? Should I have them reset or replaced? They are brand new :/

alucryd, 2023-08-25:
Oh well, after a few repairings it's finally working here.

MaasOne, 2023-10-23:
@alucryd 
What do you mean with "repairings"?
I have the same issue with "-24%" after HA restart or with "limitation maximum" is 124.

MaasOne, 2023-10-26:
OK, that's a pitty. This is definitely too much effort as a long term "solution".
Because this issue only appears after HA restart and is corrected after 1 or 2 minutes (new sensor update?) that seems to be a minor issue.

alucryd, 2023-10-30:
Well, I have some devices that never get corrected, so the only way to get them working again is to go through this procedure :/ I initially thought distance was a factor as only the farthest ones from the klf200 were affected, but now even the closest one has been stuck at `-24%` for a couple weeks, despite multiple gateway restarts.

MaasOne, 2023-11-04:
Do you have the KLF200 updated to the "actual" 2018 version (2.0.x.x)?

alucryd, 2023-11-19:
Yeah, it's fully up to date.

MaasOne, 2023-11-21:
Is your Velux (Integra) window (motor) producing date from before 2013 or after 2013?
It's just a guess...

alucryd, 2023-11-22:
I don't have any Velux device (yet, 5 blinds incoming). Right now I have a dozen Somfy IO blinds and a gateway motor paired with the gateway. They've all been installed fairly recently so I guess their production date isn't that old.

MaasOne, 2023-11-22:
Ahh, OK. Somfy is known for many issues within the io-homecontrol family. There are a lot of complaints about Somfy and io-homecontrol protocol out there.

andrazek, 2025-01-05:
Is there any update on this? The missing node issue was resolved in the latest version of pyvlx, but it didn’t address the current_position issue for the specific device, Somfy Slim Receiver IO.

wollew, 2026-02-10:
So I guess no position in the `FrameStatusRequestNotification` frames during the heartbeat either?