Julius2342/pyvlx #162: Add outdoor heater functionality?

squirblej, 2022-03-17:
Hi - I have outdoor heaters on my KLF-200. I scanned the API and they look much like lights (essentially they take a heat demand rather than a light demand). 

How easy would it be to add support for these? I'd love to get them into Home Assistant?

wollew, 2026-01-30:
I created a branch that I think could work: https://github.com/wollew/pyvlx/tree/add_outdoor_heater. So what you'll need to do is checkout this branch, create a python venv, install the dependencies (`pip install -r requirements/production.txt`), you can then run one of the [examples](https://github.com/wollew/pyvlx/blob/update_docs/examples/monitor.py) and try things like:
```
    await pyvlx.nodes['nodename'].turn_on()
    await pyvlx.nodes['nodename'].turn_off()
    await pyvlx.nodes['nodename'].set_intensity(Intensity(intensity_percent=45))
```

wollew, 2026-02-05:
Thank you for the report! So just like for lights, it seems `Intensity` is actually inverted, it's confusing but that is why on/off already worked correctly. One question: the report mentions `parameter_value="65535"` under "Error detection", I can't find this anywhere in the example log output, do you have a more complete log for that?

squirblej, 2026-02-07:
Thanks @wollew 

65535 represents a device communication failure - when there is a device registered with the KLF-200 but communication is not working

API documentation that I'm aware of is here: https://github.com/PLCHome/velux-klf200-api/blob/master/technical%20specification%20for%20klf%20200%20api.pdf

Intensity - I don't know any way of reading the actual intensity from the device using the KLF200, so the only way of knowing intensity is by visually inspecting the heater and obviously it's hard to see small differences in heat/brightness of the element. I tested off, 25% intensity, 100% intensity and could see clear differences but I don't know if there is some set of discrete values and I don't have power monitoring on it to check if different intensities draw different loads.

squirblej, 2026-02-12:
> will be included in release 0.2.29.

Fantastic. Thanks.

wollew, 2026-02-13:
Integration in HA: https://github.com/home-assistant/core/pull/162857