Julius2342/pyvlx #321: Serial Number and KLF 200

alucryd, 2023-06-22:
Hi there, it would seem the KLF 200 is not returning any serial number, resulting in cover entities in HA having no unique ID, making the whole experience very subpar as we can't (easily) customize these entities and assign them to areas.

Is that a limitation of the KLF 200? Could there be a way to work around that, or just use something else to generate a unique ID?

dumpfheimer, 2023-07-04:
hi, just stumbled upon this.
you might want to check out this custom component: https://github.com/pawlizio/my_velux
IMHO it has surpassed the development of the internal velux component by far.

alucryd, 2023-07-07:
@dumpfheimer Thank you so much for the suggestion, it looks a lot more polished indeed!