Julius2342/pyvlx #331: "Unable to send command" debug

palfrey, 2023-09-06:
So, I've been seeing a lot of the following in my Home Assistant logs recently
```
File "/usr/local/lib/python3.11/site-packages/pyvlx/opening_device.py", line 49, in set_position
    raise PyVLXException("Unable to send command")
```
which is starting to get a bit annoying. Any thoughts on how to debug this, or at least get more information than just "computer says no", because I'd love to sort this one out.

palfrey, 2023-09-06:
To clarify some details on this, this is with Home Assistant 2023.8.4. I'm pretty sure I didn't used to see this, so guessing it's an upgrade change at some point, but not sure exactly when as I was doing auto-upgrades for a while. Generally, this only happens when I'm running a scene (i.e. a bunch of velux changes in one go), although I was able to reproduce this earlier when just clicking a whole set of windows at once, so guessing it's probably to do with the gap between commands?

00-10-01-11, 2023-09-12:
I had the same issue in Home Assistant 2023.8.1 and I had to revert to 2023.7.3 which is the last working version. 

I tried every new HA version since but they have the same behavior.

palfrey, 2023-09-17:
I have a very hacky workaround with `time.sleep` filed in Core (https://github.com/home-assistant/core/pull/100523) and we'll see where that gets to...

OXERY, 2023-09-19:
I mainly see this issue when activating a scene that closes all my velux window covers. It seems that the library or klf200 is overwhelmed by the requests?
When playing around with only one window cover, it works as expected. When activating the scene mentioned before, not all window covers are closing/opening - and the ones not moving are always the same.
I was using an automation for opening and closing, which isn't working currently. Maybe I have to work around that like @palfrey  and change the automation to issue only one command at a time.

rjlee, 2023-09-27:
See also https://github.com/pawlizio/my_velux/issues/27 and https://github.com/home-assistant/core/issues/98168#issuecomment-1722253939

spitfire, 2023-11-12:
> I mainly see this issue when activating a scene that closes all my velux window covers. It seems that the library or klf200 is overwhelmed by the requests? When playing around with only one window [...]

for me they are always different. If I run my close or open scene again in HA they eventually get to the right state.

palfrey, 2023-11-23:
@Julius2342 Any thoughts on a PR to add the option for minimum delays between calls to KLF 200 devices? Like https://github.com/home-assistant/core/pull/100523, but folded into the library and redone to only add delays on subsequent rapid calls, not all of them.

pawlizio, 2023-11-24:
I implemented a semaphore to avoid parallel commands as the HA solution seems not work... Here the corresponding commit:

https://github.com/pawlizio/pyvlx/commit/70324963b0bd72835a233efc87c50584808c07d5

Julius2342, 2023-11-24:
Hmm, probably a semaphore. I would have done it quite highlevel, e.g. in ApiEvent/do_api_call . What do you think?

palfrey, 2023-11-25:
> Hmm, probably a semaphore. I would have done it quite highlevel, e.g. in ApiEvent/do_api_call . What do you think?

That seems sensible to me. https://github.com/Julius2342/pyvlx/pull/353 does this, and also solves the "why doesn't this firs call respond" issue I was seeing before by making sure we're connected before entering the Semaphore section.

PR #353 (2023-11-27): Uses a Semaphore to limit API calls to one at a time to avoid KLF issues
Fixes #331. I've tested this a few times with a set of 6 blinds, and if I do `node.close()` for all of them in parallel without this, I get roughly half closing (sometimes 2, sometimes 4-5). With this change they appear pretty stable all 6 happening. They don't quite run all at the same time, there's a bit of a gap between them (1s maybe?), but that's a lot better than them not closing.