Julius2342/pyvlx #337: Which Python versions are supported?

palfrey, 2023-09-23:
The setup.py lists 3.5 and 3.6, but the CI tests 3.8-3.11. Which are the officially supported versions?

palfrey, 2023-11-23:
@Julius2342 Any thoughts on this one?

Julius2342, 2023-11-24:
This is probably just bc nobody updated it for a long time :)

As the most common usecase is home-assistant, and there the minumum version is 3.10 I would say we change the supported versions to 3.10, 3.11 (and probably 3.9 if that does not ship with too much costs).

What do you think?

palfrey, 2023-11-24:
3.9-3.11 sounds good to me. There's some gains from not working with 3.9, but not a lot IIRC.

Julius2342, 2023-11-26:
Short update here, decided to also ditch 3.9 and go for [3.10,3.11,3,12] bc of better typehint support.

closed 2023-11-26 by Julius2342