Julius2342/pyvlx #383: Upload to Pypi

pawlizio, 2024-03-06:
Hi @Julius2342 , 

sorry have problems to update pypi for the 0.2.23. Can you please update pypi. I had to correct the discovery.py sligtly in order to allow testing at HA side for this PR https://github.com/home-assistant/core/pull/112451

Thanks and best regards!

Julius2342, 2024-03-06:
```
Writing pyvlx-0.2.23/setup.cfg
Creating tar archive
removing 'pyvlx-0.2.23' (and everything under it)
Uploading distributions to https://upload.pypi.org/legacy/
Uploading pyvlx-0.2.23.tar.gz
100% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 57.0/57.0 kB • 00:00 • 189.7 MB/s

View at:
https://pypi.org/project/pyvlx/0.2.23/
```

pawlizio, 2024-03-06:
Perfect, thanks!

closed 2024-03-06 by pawlizio

Julius2342, 2024-03-06:
The question is, why it did not work for you. Im using the makefile target and have a clean venv with `twine` installed ...