Julius2342/pyvlx #510: KLF200 becomes unavailable, if I move 2 velux connected on same KUX110

xavierforestier, 2025-04-15:
I've 2 velux connected per KUX110. 
KUX110 "A" drives Velux A1 and A2, and KUX110 "B" drives Velux B1 and B2.
If I send :
```yaml
action: cover.open_cover
target:
  entity_id:
    - cover.velux_A1
    - cover.velux_A2
    - cover.velux_B1
    - cover.velux_B2
```
Velux opens correctly, first A1 and B1, then A2 and B2. Unfortunately KL200 become then unavailable, it needs to be unplugged / replugged.

I workaround with a sequence , opens first one velux per KUX110, wait 48sec (~time to open), opens secondth velux per KUX110
```yaml
alias: Ouvrir tous les vélux
sequence:
  - action: cover.open_cover
    target:
      entity_id:
        - cover.velux_A1
        - cover.velux_B1
  - delay:
      hours: 0
      minutes: 0
      seconds: 48
      milliseconds: 0
  - action: cover.open_cover
    target:
      entity_id:
        - cover.velux_A2
        - cover.velux_B2
```
with this sequence, KLF200 stays connected !

It looks like pyvlx starts sending an "open" command to 2 velux (on same KUX110), and excepts both of them to move at the same time, unfortunately this not the way KUX110 works: One opens first, and the second waits the first one finish before starting moving . 
I think the one not moving triggers some timeout somewhere in pyvlx, that disconnect the lib (and KLF200 does not handle disconnection correctly).

pyvlx shouldn't timeout until nothing moves despite the open / close order pending: in the worst situation (all velux connected to a single KUX110) all velux will moves sequentially (in a random order) solast one may wait few minutes before finishing requested action

Julius2342, 2025-04-21:
It has been a while since i touched this area ...

Could you try to increase the timeout ... either by actively setting it within home-assistant ( `homeassistant/components/velux/__init__.py` ) :

```
  self.pyvlx = PyVLX(host=host, password=password, timeout=600)
```

or within pyvlx itself ( `old_api/pyvlx/pyvlx.py` ):

```
class PyVLX:
    """Class for PyVLX."""

    def __init__(self, path=None, host=None, password=None, timeout=600):
```

xavierforestier, 2025-05-03:
I can't reproduce anymore : everything just work fine !
Maybe I had something wrong in some automation, I don't know exactly, but it just work now. 
Tested with homeautomation-2025.4.4 which is using PyVlx-0.2.26.

You can close this issue, thanks your time and work on this module, and I'm deeply sorry for disturbing you with my wrong problem.

closed 2025-05-03 by xavierforestier