Julius2342/pyvlx #537: Adding support for setting PriorityLevel and PriorityLevelLock in GW_COMMAND_SEND_REQ

kevinior, 2025-07-13:
I'm thinking about adding support for setting the PriorityLevel and PriorityLevelLock in GW_COMMAND_SEND_REQ frames. I'd like to be able to have Home Assistant act as a fake rain sensor for my awnings, by forcing them to go up when rain is forecast soon.

In the pyvlx code I can see that `FrameCommandSendRequest` already has a field for priority, but it's always set to a default value and the priority locking fields are always sent as zeroes.

Would it be OK if I create a pull request implementing this?

Julius2342, 2025-07-15:
I'm fine with adding this ...

But: why not just send a close, when rain is awaited and do the logic within Home-assistant?

kevinior, 2025-07-15:
I want the command from Home Assistant (open, in this case) to override/block the close commands from the sun sensor.

I'm assuming that this is how our existing wind sensor works, sending an "open and priority lock" command when the wind speed is too high.